tmk/tmk_keyboard #322: Mouse BTN4/BTN5 not working in bluetooth mode

pandabird, 2016-03-31:
The buttons work fine in USB mode but does not work in bluetooth mode. Is there a fix or is this the correct behavior for bluetooth?

tmk, 2016-03-31:
I beleive you can send only 3 buttons with RN42 bluetooth module, though I cannot find clear reference in their documents. Also you cannot use horizontal wheel in BT mode.

This is limitation of the module so there is no fix.

tmk, 2016-03-31:
Just for future reference.
This is SDP info and report descriptor from RN42.

```
Parsed by http://eleccelerator.com/usbdescreqparser/

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x85, 0x01,        //   Report ID (1)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x09, 0x05,        //   Usage (0x05)
0x08,              //   Usage
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x85, 0x01,        //   Report ID (1)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x91, 0x03,        //   Output (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x65,        //   Logical Maximum (101)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0x65,        //   Usage Maximum (0x65)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x05, 0x0C,        //   Usage Page (Consumer)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x0A, 0x23, 0x02,  //   Usage (AC Home)
0x0A, 0x8A, 0x01,  //   Usage (AL Email Reader)
0x0A, 0x21, 0x02,  //   Usage (AC Search)
0x0A, 0xAE, 0x01,  //   Usage (AL Keyboard Layout)
0x0A, 0xE9, 0x00,  //   Usage (Volume Increment)
0x0A, 0xEA, 0x00,  //   Usage (Volume Decrement)
0x0A, 0xE2, 0x00,  //   Usage (Mute)
0x0A, 0xCD, 0x00,  //   Usage (Play/Pause)
0x0A, 0xB5, 0x00,  //   Usage (Scan Next Track)
0x0A, 0xB6, 0x00,  //   Usage (Scan Previous Track)
0x0A, 0xB7, 0x00,  //   Usage (Stop)
0x0A, 0xB8, 0x00,  //   Usage (Eject)
0x0A, 0x0B, 0x03,  //   Usage (0x030B)
0x00,              //   Unknown (bTag: 0x00, bType: 0x00)
0x0A, 0xB4, 0x00,  //   Usage (Rewind)
0x0A, 0xCC, 0x00,  //   Usage (Stop/Eject)
0x0A, 0x96, 0x01,  //   Usage (AL Internet Browser)
0x75, 0x01,        //   Report Size (1)
0x95, 0x10,        //   Report Count (16)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x08,        //   Usage Page (LEDs)
0x85, 0xFF,        //   Report ID (255)
[... 92 lines cut ...]
        Attribute 0x0206
                Sequence
                        Sequence
                                UINT8 0x22
                                Data 05 01 09 06 a1 01 05 07 85 01 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 85 01 19 01 29 05 91 02 95 01 75 03 91 03 95 06 75 08 15 00 25 65 05 07 19 00 29 65 81 00 c0 05 0c 09 01 a1 01 85 03 05 0c 15 00 25 01 0a 23 02 0a 8a 01 0a 21 02 0a ae 01 0a e9 00 0a ea 00 0a e2 00 0a cd 00 0a b5 00 0a b6 00 0a b7 00 0a b8 00 0a b3 00 0a b4 00 0a cc 00 0a 96 01 75 01 95 10 81 02 05 08 85 ff 95 01 75 02 09 24 09 26 81 02 75 06 81 01 c0 05 01 09 02 a1 01 09 01 a1 00 85 02 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 c0 c0 00
        Attribute 0x0207
                Sequence
                        Sequence
                                UINT16 0x0409
                                UINT16 0x0100
        Attribute 0x0208
                Bool False
        Attribute 0x0209
                Bool True
        Attribute 0x020a
                Bool True
        Attribute 0x020b
                UINT16 0x0100
        Attribute 0x020c
                UINT16 0x1f40
        Attribute 0x020d
                Bool False
        Attribute 0x020e
                Bool False
```

pandabird, 2016-03-31:
Okay, thanks!

closed 2016-03-31 by pandabird