tmk/tmk_keyboard #328: Consider open-sourcing the keymap editor

27percent, 2016-04-12:
Have you considered open-sourcing the keymap editor? I've been building my own keymap editor GUI which I'm open-sourcing, it spits out JSON data which I'm planning on sending to a server to compile, I was wondering if you were interested in open-sourcing your TMK compilation server code so I don't have to hack my own together? I'd be happy to help work on it if you do :)

tmk, 2016-04-12:
My compilation server? No serverside code. It work completely on client side, it is the beauty of my editor. All of my code is available here.
https://github.com/tmk/tmk_keyboard/tree/gh-pages/editor

No license is given obviously yet though. It uses JQuery and some other libraries I'll have to check their license but probably MIT or GPL3?

27percent, 2016-04-12:
You're kidding me! Oh wow, dude, that's crazy cool! So I'm reading over the code, am I right in thinking the firmware_hex() function in firmware.js is basically the compiled firmware minus the keymap region?

tmk, 2016-04-12:
Right. My editor just concatinates keymap definition and precompiled firmware in HEX form.

Make a contribution when you get your editor worked!

27percent, 2016-04-18:
Awesome, will do!
Would you be available to talk me through your keymap editor over email/slack? I'm working on getting a Mass Storage bootloader working on the AtMega32U4 so I need to compile firmware to bin instead of hex, it'd be great if we could get it working together!

eltang, 2016-04-19:
Someone just announced their keymap editor [here](https://www.reddit.com/r/MechanicalKeyboards/comments/4fg5hd/tmk_firmware_builder/).

tmk, 2016-04-21:
@27percent what is your hardware? Mass storage bootloader is cool.
I prefer discussion in open place! Let us know repository of your project.

27percent, 2016-04-21:
@eltang that looks cool, although it looks like it's sending a request to a server to do the compilation, that's how I was originally going to do it before I saw the amazing TMK in-browser one :)

@tmk hardware is just a pretty simple AtMega32U4 based 3x4 board, nothing fancy, although I'm almost definitely going to have to move to a different chip if I make a bigger board, there's just not enough space on the 32U4 to fit the 6k mass storage bootloader and then the TMK firmware with a bunch of layers and macros. I managed to get the Mass Storage bootloader working on it with Windows but MacOS and Linux write FAT differently than windows does, so it's going to be some more work before it's ready.

closed 2018-10-14 by tmk