tmk/tmk_keyboard #331: HHKB is slow in USB enumeration

tmk, 2016-04-16:
In Linux device enumeration process of HHKB is slower than other project like alps64. Seems like it takes 2secs pause between readout of device/product string descritpros and readout of report descriptor.

HHKB enumeration:
```
Apr 16 14:47:49 desk kernel: [346451.566281] usb 4-2: USB disconnect, device number 8
Apr 16 14:47:51 desk kernel: [346453.536045] usb 4-2: new full-speed USB device number 9 using ohci-pci
Apr 16 14:47:51 desk kernel: [346453.749024] usb 4-2: New USB device found, idVendor=feed, idProduct=cafe
Apr 16 14:47:51 desk kernel: [346453.749031] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 16 14:47:51 desk kernel: [346453.749034] usb 4-2: Product: HHKB mod
Apr 16 14:47:51 desk kernel: [346453.749036] usb 4-2: Manufacturer: t.m.k.
Apr 16 14:47:53 desk kernel: [346455.485498] input: t.m.k. HHKB mod as /devices/pci0000:00/0000:00:04.0/usb4/4-2/4-2:1.0/0003:FEED:CAFE.01AB/input/input357
Apr 16 14:47:53 desk kernel: [346455.540745] hid-generic 0003:FEED:CAFE.01AB: input,hidraw5: USB HID v1.11 Keyboard [t.m.k. HHKB mod] on usb-0000:00:04.0-2/input0
Apr 16 14:47:53 desk kernel: [346455.555554] input: t.m.k. HHKB mod as /devices/pci0000:00/0000:00:04.0/usb4/4-2/4-2:1.1/0003:FEED:CAFE.01AC/input/input358
Apr 16 14:47:53 desk kernel: [346455.608297] hid-generic 0003:FEED:CAFE.01AC: input,hidraw6: USB HID v1.11 Mouse [t.m.k. HHKB mod] on usb-0000:00:04.0-2/input1
Apr 16 14:47:53 desk kernel: [346455.632169] input: t.m.k. HHKB mod as /devices/pci0000:00/0000:00:04.0/usb4/4-2/4-2:1.2/0003:FEED:CAFE.01AD/input/input359
Apr 16 14:47:53 desk kernel: [346455.688429] hid-generic 0003:FEED:CAFE.01AD: input,hidraw7: USB HID v1.11 Device [t.m.k. HHKB mod] on usb-0000:00:04.0-2/input2
Apr 16 14:47:53 desk kernel: [346455.710026] input: t.m.k. HHKB mod as /devices/pci0000:00/0000:00:04.0/usb4/4-2/4-2:1.3/0003:FEED:CAFE.01AE/input/input360
Apr 16 14:47:53 desk kernel: [346455.800317] hid-generic 0003:FEED:CAFE.01AE: input,hidraw8: USB HID v1.11 Keyboard [t.m.k. HHKB mod] on usb-0000:00:04.0-2/input3
```

Alps64 enumeration:
```
Apr 16 14:50:18 desk kernel: [346600.620048] usb 1-1.3.4: new full-speed USB device number 49 using ehci-pci
Apr 16 14:50:18 desk kernel: [346600.714700] usb 1-1.3.4: New USB device found, idVendor=feed, idProduct=6464
Apr 16 14:50:18 desk kernel: [346600.714713] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 16 14:50:18 desk kernel: [346600.714719] usb 1-1.3.4: Product: Alps64
Apr 16 14:50:18 desk kernel: [346600.714724] usb 1-1.3.4: Manufacturer: TMK
Apr 16 14:50:18 desk kernel: [346600.718231] input: TMK Alps64 as /devices/pci0000:00/0000:00:02.1/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.0/0003:FEED:6464.01AF/input/input361
Apr 16 14:50:18 desk kernel: [346600.772359] hid-generic 0003:FEED:6464.01AF: input,hidraw1: USB HID v1.11 Keyboard [TMK Alps64] on usb-0000:00:02.1-1.3.4/input0
Apr 16 14:50:18 desk kernel: [346600.774213] input: TMK Alps64 as /devices/pci0000:00/0000:00:02.1/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.1/0003:FEED:6464.01B0/input/input362
Apr 16 14:50:18 desk kernel: [346600.828309] hid-generic 0003:FEED:6464.01B0: input,hidraw2: USB HID v1.11 Mouse [TMK Alps64] on usb-0000:00:02.1-1.3.4/input1
Apr 16 14:50:18 desk kernel: [346600.830473] input: TMK Alps64 as /devices/pci0000:00/0000:00:02.1/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.2/0003:FEED:6464.01B1/input/input363
Apr 16 14:50:18 desk kernel: [346600.884367] hid-generic 0003:FEED:6464.01B1: input,hidraw3: USB HID v1.11 Device [TMK Alps64] on usb-0000:00:02.1-1.3.4/input2
Apr 16 14:50:18 desk kernel: [346600.886281] hid-generic 0003:FEED:6464.01B2: hiddev0,hidraw4: USB HID v1.11 Device [TMK Alps64] on usb-0000:00:02.1-1.3.4/input3
```

tmk, 2016-04-17:
Boot magic at startup causes this problem, it disappears when disabling boot magic(BOOTMAGIC_ENABLE).
Matrix scan of HHKB takes more time than others but not sure why exactly. Key checking in bootmagic.c have to be optimised probably.

eltang, 2016-04-17:
I was thinking about a change to Boot Magic. How about changing the steps to look like this.

1. Plug in keyboard.
1. Hit spacebar.
1. Half a second to hit a key to configure Boot Magic.
1. Proceed to normal keyboard operation.

I want this because it can be a little difficult to hold certain Boot Magic key combinations while also plugging in a USB cable.

tmk, 2016-04-17:
I'm afraid users misconfigure their keyboard unintentionally I think we need properly difficult way to avoid it.

eltang, 2016-04-17:
I have indeed seen that happen a few times.

eltang, 2016-04-18:
Instead of requiring the spacebar to be pressed, how about requiring (a) special keyword(s) to be typed?

eltang, 2016-04-21:
I just saw that QMK implemented Magic keys with dedicated keycodes. What do you think of that? Users who don't want to use Magic can simply leave those keys out of their keymaps.

tmk, 2016-04-21:
Nice, I like to add it in my TODO list.
Or you can create new issues here to let us know nice features from QMK or any ohter repos.

Maybe it is not related to this issue.

eltang, 2016-04-21:
I see. Should I create an issue with a running list of features that I think might be worth adding to QMK?

tmk, 2016-04-21:
Ah, yes one issue would be better to list QMK features.

eltang, 2016-04-21:
Cool. One issue coming up

closed 2020-12-10 by tmk