tmk/tmk_keyboard #332: How Media keys works

tmk, 2016-04-17:
# USB LUFA
## Linux Xorg(Ubuntu/Xfce):

Keycode|Action
-----|-----
PWR|Pop out Logout dialog
SLEP|No action??
WAKE|No action(XF86WakeUp)
MUTE|Mute
VOLU|Volume Down
VOLD|Volume Up
MNXT|works in VLC media player (XF86AudioNext)
MPRV|works in VLC media player (XF86AudioPrev)
MFFD|?? (XF86AudioForward)
MRWD|?? (XF86AudioRewind)
MSTP|works in VLC media player (XF86AudioStop)
MPLY|works in VLC media player (XF86AudioPlay)
EJCT|No action(NoSymbol)
MSEL|No action(XF86Tools)
MAIL|No action(XF86Mail)
CALC|No action(XF86Calculator)
MYCM|No action(XF86Explorer)
WSCH|Load Google in Chrome49 (XF86Search)
WHOM|Load home page in Chrome49 (XF86HomePage)
WBAK|History back in Chrome49 (XF86Back)
WFWD|History forward in Chrome49 (XF86Forward)
WSTP|??(Cancel)
WREF|Refresh in Chrome49 (XF86Reload)
WFAV|Toggle bookmark bar in Chrome49 (XF86Favorites)

## Windows 10
Keycode|Action
-----|-----
PWR|Sleep with my ThinkPad
SLEP|Sleep with my ThinkPad
WAKE|No action
MUTE|Mute
VOLU|Volume Down
VOLD|Volume Up
MNXT|works in Windows media player
MPRV|works in Windows media player
MFFD|??
MRWD|??
MSTP|works in Widnows media player
MPLY|works in Windows media player
EJCT|No action
MSEL|Open media player
MAIL|Open default Mailer
CALC|Open Calculator
MYCM|Open Explorer
WSCH|Open Search Windows dialog
WHOM|Load home page in Internet Explorer
WBAK|History back in Internet Explorer
WFWD|History forward in Internet Explorer
WSTP|Stop loading in Internet Explorer
WREF|Refresh in Internet Explorer
WFAV|Toggle bookmark pane in Internet Explorer

## OSX

# Bluetooth HHKB w/ RN42
## Linux Xorg(Ubuntu/Xfce)

## Windows 10

## OSX

shelaf, 2016-07-09:
I tested on OSX.
Environment: 
Keyboard: HHKB Pro2 (with Hasu's Alternative contoller)
Firmware: tmk_keyboard (develop branch, BUILD: f8bab8c)
OS: El Capitan(10.11.5)

# Protocol: USB LUFA
## OSX (El Capitan)

Keycode | Action
--- | ---
PWR | press and release: sleep / hold down: show restart, sleep, or shut down dialog
SLEP | press and release: sleep / hold down: show restart, sleep, or shut down dialog
WAKE | press and release: sleep / hold down: show restart, sleep, or shut down dialog
MUTE | Mute
VOLU | Volume Up
VOLD | Volume Down
MNXT | Next track. works in VLC, VOX
MPRV | Previous track. works in VLC, VOX
MFFD | Next track or fast-foward(hold down). works in VLC, VOX
MRWD | Previous track or rewind(hold down). works in VLC, VOX 
MSTP | No action
MPLY | works in VLC, VOX
EJCT | No action
MSEL | No action
MAIL | No action
CALC | No action
MYCM | No action
WSCH | No action
WHOM | No action
WBAK | No action
WFWD | No action
WSTP | No action
WREF | No action
WFAV | No action

tmk, 2016-07-09:
Thank you. It is really helpful.

tmk, 2019-01-28:
Report descriptor fix(feebc23) for #435 may affect to this.
See qmk/qmk_firmware#963 also for detail.

closed 2021-02-08 by tmk

CJacobStucki, 2021-03-09:
MPLY doesn't seem to be working in Windows 10 (MPRV and MNXT work fine).  

Backround;
I am using a kbparadise v60 type R polestar, and have programmed it using keyboard layout editor and the tgz.io tool.  I use the eep file with a script from kbparadise to flash the keyboard.  The .eep file doesn't provide any insite, but the optional .c file from tgz.io does, and it seems to list the key as MPLY correctly.  After flashing however, that keycode does nothing...

Can you point me towards a fix for this?  I would greatly appreciate it :)

(I've attached the .c file as a .txt file because github doesn't support attaching .c files for some reason...)

[keymap_c_file.txt](https://github.com/tmk/tmk_keyboard/files/6106193/keymap_c_file.txt)

tmk, 2021-03-09:
I just confirmed `KC_MPLY`(`KC_MEDIA_PLAY_PAUSE`) still works on Ubuntu/Xfce and Windows 10 with the latest code(a93677cb).

Contact with proper firmware maintainer for your keyboard.