tmk/tmk_keyboard #336: Adding features from QMK

eltang, 2016-04-21:
Here is a list of features from QMK that I think would be worthwhile to incorporate into TMK. I'll keep this list updated as I see new features being developed there.

1. Ability to define key matrix with pin names
1. Magic keycodes
1. WS2812 RGB light effects
1. Scripts for setting up build environment
1. A cache that prevents keys from getting stuck when switching layers

If anyone has any questions about these features, I can try to answer them.

tmk, 2016-04-21:
Yeah, I think this is most wanted and I'll like to make it as default.
- Prevent stuck modifiers https://github.com/jackhumbert/qmk_firmware/pull/182

eltang, 2016-04-21:
I actually wrote the final encoding for that pull request. I trying to refactor the code right now, but `layer_switch_get_action` is called not only by `process_action` but also `process_tapping` and `is_tap_key`. You wrote the tapping code, so do you think it is appropriate to use the cache in the latter two cases?

tmk, 2016-04-21:
@yashikno is not related to me and this project, but github displays his avtar on my old commits. Becuase I've used  'noname@nowhere' as email address when making commits before, but then he registered @yashikno on github.com with 'noname@nowhere'. Not his fault, maybe github's.

Yeah, tapping code is mess and I also don't want to thouch :)
I'll look into those cases later and cherry-pick from QMK.

eltang, 2016-04-21:
Okay. Do you want me to send you the code I have so far (to a new branch)?

eltang, 2016-04-21:
Also, do you have plans to clean up the tapping code? I have tried and failed a few times to understand it.

tmk, 2016-04-21:
Yes, I like to refactor it, the current code is result of heuristic tunes to my typing and set of  ad hoc fixes. It  is very difficut to understand it even for me.

I remember 'touch cursor'  tool has very clean code and works well. I would refer  to its code If I refactor tapping.
http://martin-stone.github.io/touchcursor/

eltang, 2016-04-21:
Okay. Maybe we should delay the integration of the cache until after the tapping code has been refactored. We should have a clearer idea of where the cache should be used then.

eltang, 2016-04-21:
Could you please point me to the relevant part of the code? I can't seem to find it.

eltang, 2016-04-26:
@tmk Do you have a timeline for implementing this feature?

tmk, 2016-04-27:
I have no definite plan about jackhumbert#182 so far. But I'll try if I have time enough.

tmk, 2016-04-27:
Or you mean refactor of tapping code? hmm, I don't know, but it won't basically as far as it works for me.

eltang, 2016-04-27:
Those two changes might have to be done simultaneously, don't you think? Look back at our previous discussion.

tmk, 2016-04-27:
If needed I'll try to rewrite. I did't look the patch closer but I thought tapping is no relation. I missed something?

eltang, 2016-04-27:
The tapping code calls `layer_switch_get_action` in multiple places and the cache may or may not be needed in those instances.

tmk, 2016-04-27:
could you give me a link to the source code?

eltang, 2016-04-27:
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action.c#L67
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_tapping.c#L118

tmk, 2016-04-27:
ah, `store_or_get_action()` doesn't work with tapping? why?

eltang, 2016-04-27:
I'm not actually sure whether or not it works. I just thought it might be good to keep in mind that the tapping code calls `layer_switch_get_action`. `store_or_get_action` is really only different from `layer_switch_get_action` when the layer is changed while a key is held down.

tmk, 2016-04-27:
I don't see any reason it doesn't work but I may be wrong.
As you refered QMK doesn't use `store_or_get_action()` in those place for some reason.

Thanks for headsup!

eltang, 2016-04-27:
All right. Best of luck to you in getting those changes made. I'm looking forward to what you come up with. Vifon/qmk_firmware#2 contains some unfinished changes to the cache code that I made.

eltang, 2016-04-28:
Okay, so I have a question. Are `ACT_MOUSKEY` and `ACT_USAGE` still used in `action_code.h`? I don't think QMK's special keycodes are implemented in the best way possible and I was wondering if it was possible to make them into action codes.

eltang, 2016-05-07:
@tmk How's this coming along? I just had another idea for a great feature that resulted from a discussion over at the QMK repo. I think that when an `ACTION_MACRO_TAP` macro times out, the macro should be called a final time with `record->tap.end`. What do you think? I believe it will be very easy to implement.

timofonic, 2016-10-06:
Is TMK MIA? :(

closed 2017-05-30 by tmk