tmk/tmk_keyboard #338: Feather Bluefruit Cortex M0

sfcgeorge, 2016-04-23:
Hi, great firmware, thank you!

I came across the [Adafruit Feather Bluefruit](https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le/hidkeyboard) which has a nice Cortex M0 chip, Bluetooth LE and a good number of IO. I'd love to build a keyboard with it.

Does TMK support M0?

The Bluetooth stack has a [built in keyboard mode](https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le/hidkeyboard) that works with computers, iOS and Android, could that be hooked up to TMK somehow?

tmk, 2016-04-24:
Hi,
Coretx M0 can supported with module chibios or mbed under 
tmk_core/protocol/. I don't know they have support for the exact chip on 
the board but it won't be so diffcult to add support if needed.

You can use tmk_core to make a keyboard with the board.

kejadlen, 2016-04-24:
There's also the [Adafruit Feather 32u4 Bluefruit](https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/overview), which might be easier to use with tmk_keyboard.

flabbergast, 2016-04-25:
The Feather Bluefruit's main MCU is ATSAMD21G18, which is not supported by ChibiOS (yet). I mean the "Cortex M0 core" bit is OK, but chibios also includes HAL which is used by TMK {this means things like GPIO and USB support} and that differs from chip to chip. The HAL layer for ATMEL's ARM MCUs is not there.
I don't know about MBED.

closed 2016-05-13 by tmk