tmk/tmk_keyboard #339: terminal_bluefruit still working?

esell, 2016-05-04:
I am trying to build a bluetooth enabled firmware for the adafruit ez-key using the terminal_bluefruit as a starting point. The issue is that I can't get it to compile before I even start customizing it, below is the output:

```
mkdir -p obj_terminal_bluefruit/protocol/bluefruit
Compiling C: ../../tmk_core/protocol/bluefruit/bluefruit.c
avr-gcc -c -mmcu=atmega32u4        -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=4096 -DPS2_USE_INT -DPROTOCOL_BLUEFRUIT -DPROTOCOL_PJRC -DPS2_USE_INT -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DVERSION=7613efc-dirty -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_terminal_bluefruit/protocol/bluefruit/bluefruit.lst -I. -I../../tmk_core -I../../tmk_core/protocol -I../../tmk_core/protocol/bluefruit -I../../tmk_core/protocol/pjrc -I../../tmk_core/protocol -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_terminal_bluefruit_protocol_bluefruit_bluefruit.o.d  ../../tmk_core/protocol/bluefruit/bluefruit.c -o obj_terminal_bluefruit/protocol/bluefruit/bluefruit.o
../../tmk_core/protocol/bluefruit/bluefruit.c:46:13: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 static void bluefruit_trace_header()
             ^
../../tmk_core/protocol/bluefruit/bluefruit.c:53:13: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 static void bluefruit_trace_footer()
             ^
../../tmk_core/protocol/bluefruit/bluefruit.c: In function 'send_consumer':
../../tmk_core/protocol/bluefruit/bluefruit.c:161:15: error: 'AL_CC_CONFIG' undeclared (first use in this function)
     (usage == AL_CC_CONFIG         ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:161:15: note: each undeclared identifier is reported only once for each function it appears in
     (usage == AL_CC_CONFIG         ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:162:15: error: 'AL_EMAIL' undeclared (first use in this function)
     (usage == AL_EMAIL             ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:163:15: error: 'AL_CALCULATOR' undeclared (first use in this function)
     (usage == AL_CALCULATOR        ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:164:15: error: 'AL_LOCAL_BROWSER' undeclared (first use in this function)
     (usage == AL_LOCAL_BROWSER     ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:165:15: error: 'AC_SEARCH' undeclared (first use in this function)
     (usage == AC_SEARCH            ? 0x0400  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:166:15: error: 'AC_HOME' undeclared (first use in this function)
     (usage == AC_HOME              ? 0x0100  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:167:15: error: 'AC_BACK' undeclared (first use in this function)
     (usage == AC_BACK              ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:168:15: error: 'AC_FORWARD' undeclared (first use in this function)
     (usage == AC_FORWARD           ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:169:15: error: 'AC_STOP' undeclared (first use in this function)
     (usage == AC_STOP              ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:170:15: error: 'AC_REFRESH' undeclared (first use in this function)
     (usage == AC_REFRESH           ? 0x0000  : \
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
../../tmk_core/protocol/bluefruit/bluefruit.c:171:15: error: 'AC_BOOKMARKS' undeclared (first use in this function)
     (usage == AC_BOOKMARKS         ? 0x0000  : 0)))))))))))))))))))
               ^
../../tmk_core/protocol/bluefruit/bluefruit.c:179:23: note: in expansion of macro 'CONSUMER2BLUEFRUIT'
     uint16_t bitmap = CONSUMER2BLUEFRUIT(data);
                       ^
make: *** [obj_terminal_bluefruit/protocol/bluefruit/bluefruit.o] Error 1
```

Is there a better way to enable bluetooth support via the adafruit ez-key?

esell, 2016-05-04:
As an FYI I am working off of a branch whose last commit was 7613efca999f1b0b90c57cedfa67eda1161b8d44

fredizzimo, 2016-05-05:
Hi,

I don't know anything about it, but I did some investigation.

The warnings like ```warning: function declaration isn't a prototype [-Wstrict-prototypes]
 static void bluefruit_trace_header()```, have probably always been there, as the c language would need ```static void bluefruit_trace_header(void)```, and the functions have remained unchanged since they were added.

The errors are caused by the action map support in https://github.com/tmk/tmk_keyboard/commit/5e9b21d77d71bede3fbcbefd7567af22f3b0d153. The macros have changed names like AL_CC_CONFIG -> APPLAUNCH_CC_CONFIG. See the actual commit diff for the complete list.

The iwrap keyboards with the EXTRAKEY_ENABLE macro defined seems to have the same problem, so it appears like @tmk for some reason didn't do a search and replace when he changed the macro names. 

I suggest that you fix at least the Bluefruit version and send a pull request, but it's probably worth fixing the iwrap version as well. I could do it myself, but I have other things to do right now.

BTW, I also noticed that this action map support hasn't made it's way into the tmk_core library master, or even the development branch yet, even if it's in use by tmk_keyboard. @tmk, what's the reason for that? I think the tmk_core should always be at least as new as the version used by tmk_keyboard.

esell, 2016-05-05:
That did it, thanks for the tips, I'll submit a PR shortly with the changes

tmk, 2016-05-05:
@fredizzimo , nice catch!

BTW, I want to include change #318 with actionmap but it breaks existent keymap files so I'm procrastinating to merge into tmk_core repo.

tmk_core repo may takes changes from tmk_keyboard/tmk_core as possible if I think it is safe and I have time to do. tmk_keyboard/tmk_core will be changed first and the latest code, at this time at least.

closed 2016-05-05 by tmk