tmk/tmk_keyboard #340: Treat warnings as errors

fredizzimo, 2016-05-05:
It would be nice, if all the warnings are fixed, and the warnings are treated as errors. As a C++ programmer I have very little C experience, most from adding TMK support for the Infinity Ergodox. However from the little experience I have, I find that most C warnings should really be errors. For example it lets say that you initialize a struct of function pointers, with functions of the wrong signature with just this warning ``` warning: initialization from incompatible pointer type```. If you manage to miss that warning from output, your code will very likely not work.

Other cases are a little bit better, as you get a linker error for something that should have been a compilation error.

NOTE: there's a separate bug for fixing some warnings https://github.com/tmk/tmk_keyboard/issues/317.

tmk, 2016-05-06:
It would be nice practice but I'd like to leave it to choice of project owners. Each developer have own preference on this topic. I think it is enough if they are able to add `-Werror` to compile option in project's Makefile, if not it should be fixed in build scripts of tmk_core.

I for one prefer to iterate of compile and test loop in fast cycles with ignoring warnings if possible. But I know warnings should be treated in the end but ... people are lazy..l you know.

fredizzimo, 2016-05-06:
Personally I have seen so many bugs being detected by high warning levels (both in C as explained above, but also with C++), that I would never recommend anyone to not treat warnings as errors. 

Also if some projects can ignore the warnings, you will start to get warnings in the core library. Say people from some project with disabled warnings contribute to the core project. Other projects then have to fix those warnings before they can continue (or relax their own warning policy), even if it's not their fault.

Project specific warnings also reduces the usefulness of  something like building all known keyboard projects in a single command and see that they are compatible with the changes you are making. Note that this is not possible yet, but I think we should add support for it and make it being run on a continuous integration server. 

Anyway I don't think it's possible to add ```-Werror``` option to your keyboard project at the moment, as the compiler flags are not exposed and customizable. So some additional changes would be needed to support that.

There's another way to support the fast iteration that you are talking about (although I personally don't think it's faster to detect that the keyboard is crashing at run-time, than to see an error immediately at compile time). Just add an option to the make command itself, like this for example ```make WARN_AS_ERROR=no```("yes" would be the default). Then you just remove the flag when testing for the final time before committing the changes, or at the very latest before pushing them. This would also of course require some small changes to the make files. 

If it's done that way, you could still support what you have suggested, and add the option to the project specific makefile, which would change the default to no(don't treat warnings as errors). In that case the option could still be overridden by the continuous integration server, which should always set it to yes. I don't recommend doing it that way though, since then we still have inconsistencies between different projects, and we don't have an CIS yet for verifying different projects. 

We could also have a global default of "no". But at leas as long as we don't have a CIS, which verifies all the warnings I don't recommend that either.