tmk/tmk_keyboard #346: keyboard and physical trackpoint

acornejo, 2016-05-13:
Does the TMK firmware (running on AVR) can be made to support both a keyboard and a physical trackpoint?

I.e. To build a keyboard + trackpoint combo, do I need to have two separate USB outputs and a USB hub, or is it possible to have the same chip drive both?

eltang, 2016-05-13:
TMK can currently emulate a keyboard and a mouse at the same time. You will need to find a way to convert signals from the TrackPoint to mouse reports.

acornejo, 2016-05-13:
I was hoping to simply connect the ps2 signal to the processor, and then have the firmware send that through the USB port together with any keyboard presses. However, not having worked with PS2 or TMK before, I am not sure if that is feasible.

tmk, 2016-05-13:
I'm not sure what you expect with the word 'support'. Trackpoint is just a PS/2 mouse so it works.
See onekey project. Enable PS2_MOUSE_ENABLE and {PS2_USE_INT or PS2_USE_USART} in Makefile and configure pins for trackpoint in config.h, it should work.

And see this for reset.
https://github.com/tmk/tmk_keyboard/wiki/FAQ#trackpoint-needs-reset-circuitps2-mouse-support

acornejo, 2016-05-13:
thanks a lot @tmk. that was poor wording on my part, by support I simply meant "possible without having to write my own firmware from scratch". From what you say it seems that this project pretty much does all that I need, I just need to build it and wire it appropriately.

I'll buy the trackpoint and tmk supported keyboard on ebay and give it a try. Feel free to close this bug, I'll post comments here if I encountary any problems.

gzeronet, 2016-05-15:
Hi, there. @acornejo just let you know, it can work well. Actually, I've already made one for myself, trackpoint(nmb E420 from taobao.com) + teensy(2.0) + tmk_keyboard(customed gh60). But I can't let the mouse buttons(pins on trackpoint) work now. Still find solution...

acornejo, 2016-05-15:
@gzeronet that is great news.

I plan to have a very similar setup; I just ordered the kc60 from massdrop (which is based of gh60) https://www.massdrop.com/buy/kc60-mechanical-keyboard, and ordered a used T40 keyboard with trackpoint from ebay. Sadly, it will be a month or two before they get here.

Btw, do you have a blog or photos that document how you drilled the PCB and what did you do to extend the trackpoint stick above the keys? The best solution I've seen so far is to use epoxy to glue a nut to the trackpoint and use a flat head screw as the trackpoint stick.

gzeronet, 2016-05-15:
@acornejo , well, I just finish the prototype (based on cherry G84-4100),  and plan to submit the project to github after I make a real custom gh60 keyboard. Currently, I just drill a hole on the 4100 PCB，and add plastic pipe （from ballpoint pen） as trackpoint stick. As a prototype you know. Would like to share something if that can help.

gzeronet, 2016-05-15:
https://i.imgsafe.org/a0943b1.jpg

closed 2016-05-31 by tmk