tmk/tmk_keyboard #352: Apple adjustable keyboard 1993 does not work over ADB cord. converter-usb_converter

columna1, 2016-05-31:
I had made the adapter and soldered directly to the internal wires and it worked fine (although somewhat intermittent when moved the wrong way) it had no pullup resistor and seemed to work. I recently got some ADB mice and decided to make the adapter external so I ripped a female adapter off of an old tv board and soldered everything up with a pullup resistor checking MANY times that the pins are correct. Now the keyboard does not work Error -30 and the mouse flies up to the top left, but the mice work perfectly fine without a problem. I used a 10k 1/4wat resistor. would there be any reason that this problem exists?

columna1, 2016-05-31:
according to the information I found error -30 just means it can't communicate. The same error occurs when nothing is connected

tmk, 2016-05-31:
right, no connection or hardware error.

columna1, 2016-05-31:
then why was it working before and not now. we changed absolutely nothing except for they way they connected.

tmk, 2016-05-31:
I think your hardware was iffy from the begining.

> I had made the adapter and soldered directly to the internal wires and it worked fine (although somewhat intermittent when moved the wrong way)

columna1, 2016-05-31:
I had presumed it was because it was missing the pullup resistor. other than that it worked perfectly fine. it would give a -30 error if the wires from the keyboard to the teensy were twisted in the wrong way or something similar