tmk/tmk_keyboard #354: Stm32F103 ChibiOS HardFault during reboot

obuhz, 2016-06-02:
I tried keyboard/stm32_f103_onekey on my generic Stm32f103c8t6 board. 
Everything worked flawless but after reboot the OS has a high chance of not recognizing the keyboard anymore (usb descriptor read error).

With some exploration, I found the MCU encountered a HardFault during OS boot. The stack dump for this HardFault showed something like this:
```
ctx = {r0 = 0x20000868 <USBD1>, r1 = 0x5, r2 = 0x20000a58 <__compound_literal.0>, r3 = 0x10, r12 = 0x200004f8, lr_thd = 0x0, pc = 0x80033cc <usbStartTransmitI+28>, xpsr = 0x41000000}
```

Normally a HF is caused by program logic. After some further investigation, I found by commenting out line 154 and 156 line in main.h could solve this issue. 
```
diff --git a/tmk_core/protocol/chibios/main.c b/tmk_core/protocol/chibios/main.c
index 9c5254f..ff0d01b 100644
--- a/tmk_core/protocol/chibios/main.c
+++ b/tmk_core/protocol/chibios/main.c
@@ -151,9 +151,9 @@ int main(void) {
       }
       /* Woken up */
       // variables have been already cleared
-      send_keyboard_report();
+//      send_keyboard_report();
 #ifdef MOUSEKEY_ENABLE
-      mousekey_send();
+//      mousekey_send();
 #endif /* MOUSEKEY_ENABLE */
     }
```

I don't have enough knowledge about USB spec or ChibiOS so the root cause of this issue is still unknown to me.

tmk, 2016-06-02:
Great skill to trace a problem! I guess you are using debugger and it seems I have to learn it finally instead of printf()  :)

I guess `send_keyboard_report()` was called when USB is not ready and we have to check its state more. I believe @flabbergast can speculate this way better.

flabbergast, 2016-06-02:
That is somewhat strange - that code should be only called when the computer wakes up from suspend. If it does get called, the TMK firmware is in a state in which it thinks everything has been already enumerated (which is not correct, obviously, since you've rebooted).

There are a couple of things that we can try to do about this: add guards for USB_DRIVER.state (i.e. wait until it is USB_ACTIVE) around the calls; or a delay (which I don't really like).

@obuhz If you have a debugger, could you figure out the value of USB_DRIVER.state when the hard fault occurs?
{You can do that by adding [this file](https://github.com/flabbergast/chibios-projects/blob/master/projects/testing/hardfault.c) to your keyboard dir, add `hardfault.c` to the sources `SRC` in the `Makefile`. It makes the MCU break into the debugger whenever a hard fault occurs. You can then examine the RAM contents.}

obuhz, 2016-06-03:
flabbergast, I added your file  to the SRC, but my compiler (arm-none-eabi-gcc) complained with the following error:
```
Compiling hardfault.c
hardfault.c: In function 'HardFault_HandlerC':
hardfault.c:53:32: warning: variable '_MMAR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _MMAR ;
                                ^
hardfault.c:52:32: warning: variable '_BFAR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _BFAR ;
                                ^
hardfault.c:51:32: warning: variable '_AFSR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _AFSR ;
                                ^
hardfault.c:50:32: warning: variable '_DFSR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _DFSR ;
                                ^
hardfault.c:49:32: warning: variable '_HFSR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _HFSR ;                                ^
hardfault.c:48:32: warning: variable '_CFSR' set but not used [-Wunused-but-set-variable]
         volatile unsigned long _CFSR ;
                                ^
hardfault.c:47:32: warning: variable 'stacked_psr' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_psr ;
                                ^
hardfault.c:46:32: warning: variable 'stacked_pc' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_pc ;
                                ^
hardfault.c:45:32: warning: variable 'stacked_lr' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_lr ;
                                ^
hardfault.c:44:32: warning: variable 'stacked_r12' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_r12 ;
                                ^
hardfault.c:43:32: warning: variable 'stacked_r3' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_r3 ;
                                ^
hardfault.c:42:32: warning: variable 'stacked_r2' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_r2 ;
                                ^
hardfault.c:41:32: warning: variable 'stacked_r1' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_r1 ;
                                ^
hardfault.c:40:32: warning: variable 'stacked_r0' set but not used [-Wunused-but-set-variable]
         volatile unsigned long stacked_r0 ;
                                ^
/tmp/cchniJVb.s: Assembler messages:
/tmp/cchniJVb.s:75: Error: lo register required -- `ldmia r0,{ip,lr}'
/tmp/cchniJVb.s:94: Error: lo register required -- `str ip,[sp]'
/tmp/cchniJVb.s:96: Error: lo register required -- `str lr,[sp,#4]'
/tmp/cchniJVb.s:112: Error: instruction not supported in Thumb16 mode -- `adds r3,r3,#16'
/tmp/cchniJVb.s:116: Error: invalid offset, value too big (0xFFFFFFF4)
/tmp/cchniJVb.s:119: Error: invalid offset, value too big (0xFFFFFFF8)
/tmp/cchniJVb.s:125: Error: invalid offset, value too big (0xFFFFFFFC)
../../tmk_core/tool/chibios/chibios/os/common/startup/ARMCMx/compilers/GCC/rules.mk:212: recipe for target 'build/obj/hardfault.o' failed
make: *** [build/obj/hardfault.o] Error 1
```

This program, as I understand it, is to perserve HF stack and several register values, I will try to see if I can get those.

BTW, the debugger I use is a copycat st-link v2.1 I bought several years ago from Taobao. I also use texane/stlink and arm-none-eabi-gdb to do the debug work.

obuhz, 2016-06-03:
I rewrote the hardfault.c and got these
```
Program received signal SIGTRAP, Trace/breakpoint trap.
HardFault_Handler () at hardfault.c:42
42          bkpt();
(gdb) i loc
ctx = {r0 = 0x20000860 <USBD1>, r1 = 0x1, r2 = 0x20000a38 <__compound_literal.0>, r3 = 0x8, r12 = 0x200004f0, 
  lr_thd = 0x0, pc = 0x8002bdc <usbStartTransmitI+28>, xpsr = 0x41000000}
_CFSR = 16778240
_HFSR = 1073741824
_DFSR = 9
_AFSR = 0
_BFAR = 3758157112
_MMAR = 3758157108
```

My hardfault.c 
```C
#include <string.h>
#include <ch.h>
/**
 * Executes the BKPT instruction that causes the debugger to stop.
 * If no debugger is attached, this will be ignored
 */
#define bkpt() __asm volatile("BKPT #0\n")

void HardFault_Handler(void) {                                                                                   
    //Copy to local variables (not pointers) to allow GDB "i loc" to directly show the info
    struct port_extctx ctx;
    volatile unsigned long _CFSR ;
    volatile unsigned long _HFSR ;
    volatile unsigned long _DFSR ;
    volatile unsigned long _AFSR ;
    volatile unsigned long _BFAR ;
    volatile unsigned long _MMAR ;
    //Get thread context. Contains main registers including PC and LR
    memcpy(&ctx, (void*)__get_PSP(), sizeof(struct port_extctx));
    (void)ctx;

    // Configurable Fault Status Register
    // Consists of MMSR, BFSR and UFSR
    _CFSR = (*((volatile unsigned long *)(0xE000ED28))) ;   

    // Hard Fault Status Register
    _HFSR = (*((volatile unsigned long *)(0xE000ED2C))) ;

    // Debug Fault Status Register
    _DFSR = (*((volatile unsigned long *)(0xE000ED30))) ;

    // Auxiliary Fault Status Register
    _AFSR = (*((volatile unsigned long *)(0xE000ED3C))) ;

    // Read the Fault Address Registers. These may not contain valid values.
    // Check BFARVALID/MMARVALID to see if they are valid values
    // MemManage Fault Address Register
    _MMAR = (*((volatile unsigned long *)(0xE000ED34))) ;
    // Bus Fault Address Register
    _BFAR = (*((volatile unsigned long *)(0xE000ED38))) ;
    //Cause debugger to stop. Ignored if no debugger is attached
    bkpt();
    NVIC_SystemReset();
}
```

I don't think this is caused by USBSTATE because in send_keyboard() func, you already checked it.

flabbergast, 2016-06-03:
Right, sorry, should have noticed you already had the context in the first report. Thanks for fixing the hardfault.c code - it is indeed just to preserve register values.

And you're also correct that the USB_DRIVER.state is checked in send_keyboard(), so the state is already USB_ACTIVE.

From the registers you reported above, the HardFault is caused by an unaligned read/write - the code in usb transmit in the chibios USB driver does do a 16bit "word" reads from the buffer it's transmitting from, and these need to be aligned. We've had a very similar problems on startup, hence [this](https://github.com/tmk/tmk_core/blob/master/protocol/chibios/main.c#L123-L128). As far as I can tell, the reason was that the chibios internal USB state machine state changes to USB_ACTIVE a little *before* the USB structure points to the right places, and so if there's a call to usbStartTransmitI too early, the pointer to the "tx_buffer" points to a random garbage place which is often not 16bit aligned, hence the HardFault. I think this is a problem with the chibios USB driver, not the TMK code per se.

Could you try the same workaround as for the start-up hard faults, i.e. a delay of about 50ms?

obuhz, 2016-06-06:
I added 50ms delay to the main.c and downloaded it to a real keyboard and used for 2 days. So far everything is right.
```
diff --git a/tmk_core/protocol/chibios/main.c b/tmk_core/protocol/chibios/main.c
index 9c5254f..27742ea 100644
--- a/tmk_core/protocol/chibios/main.c
+++ b/tmk_core/protocol/chibios/main.c
@@ -151,6 +151,7 @@ int main(void) {
       }
       /* Woken up */
       // variables have been already cleared
+      chThdSleepMilliseconds(50);
       send_keyboard_report();
 #ifdef MOUSEKEY_ENABLE
       mousekey_send();
```