tmk/tmk_keyboard #356: Matrix.c pin out guide?

jasonm23, 2016-06-06:
I'm trying to make a handwire firmware and while I could just rewire my teensy to match the Phantom pinouts I'd like to solve the problem with software.

```
 //* row: 0   1   2   3   4   5
 //* pin: B0  B1  B2  B3  B7  D0

 //* col: 0   1   2   3   4   5   6   7   8   9   10  11  12  13  14  15  16
 //* pin: F0  F1  F2  F3  F4  F5  F6  F7  B6  B5  B4  D7  C7  C6  D3  D2  D1
```

I have no idea how to calculate the DDRx PORTx PINx I'll need for init_rows, scan_columns etc.

How is it done?

jasonm23, 2016-06-06:
My non functioning fork is here https://github.com/jasonm23/tmk_keyboard/blob/handwired-tkl-ocodo/keyboard/handwired-tkl-ocodo/matrix.c

TD22057, 2016-06-06:
FYI - I have a c++ class that takes care of this for you.  You can basically say something like:

TeensyPin rows[] = { TPIN(B,0), TPIN(B,1), ... }

and then use readable code to toggle the pins, check their values, etc.  It's slower than doing bit manipulation but who cares - it's keyboard and it still runs at ~1000Hz.  For me, it makes the matrix code trivial to write, it's much easier to read, and there is no debugging necessary.  Basically once the row and col arrays are defined, the exact same matrix code is used for all my keyboards.  I'll upload it to github later tonight and you can take a look.

jasonm23, 2016-06-06:
That sounds great, look forward to it.

TD22057, 2016-06-07:
Here it is: https://github.com/TD22057/tmk_keyboard/tree/shark/keyboard/shark
Take a look at matrix.cpp - the cpp extension is required since it's c++ code and there are a few extern "C" statements I had to add to make it work.  If you copy TeensyPin.hpp and matrix.cpp and edit matrix.cpp it should be much simpler having to deal with bit/port manipulations.  You can use my debouncing system or TMK's - it's up to you.

jasonm23, 2016-06-07:
I'll take a look, much appreciated!

jasonm23, 2016-06-07:
I like the code, should be easy to use.  Thanks.

jasonm23, 2016-06-09:
@TD22057  Hey Ted, thanks again, your project has been amazingly helpful.  I should probably open this question in a new issue, but I figured I'd just ask you for help quickly, to see if you know what I've done wrong.

In a nutshell, all is working correctly, except columns 15 & 16.

Please note, I've triple checked the hardward build (swapped the column hook ups to make sure) and the teensy is connected cleanly. (these cols are on D2(col15) and D1(col16))

All keys on Col15 (except for UP, which has no right neighbor) will fire their own key code, but also right adjacent Col16 key. All Col16 switches do no register any key codes when closed.

Any ideas?  Here's my code:

https://github.com/jasonm23/tmk_keyboard/tree/handwired-tkl-ocodo/keyboard/ocodo-handwired-tkl-2

TD22057, 2016-06-09:
Code looks fine to me.  When you swapped the columns, did those columns start working?  If you see the same behavior on that column, then it's a wiring problem.  If you swap columns around and whatever column is attached to D1 and D2 is weird, then I don't know - I'd guess there is a short on the teensy or somewhere else that would be causing that.  Uncomment the DBG_KEYPRESS macro and run the teenys utility hid_listen.  That will print key up/down messages for everything it sees which can help.  You can also try using some hook up wire to those two pins - run w/ dbg and touch the wires to ground to see if it generates a press event.  That should tell whether it's the pins, the code, or the matrix.

jasonm23, 2016-06-09:
Thanks Ted, I'll try again this evening.

I tried debug from magic yesterday, and basically the two columns acted as Col 15 when I swapped the 15/16 column from the teensy.

Both with the same action of triggering their col16 neighbor on press.

I also tried directly shorting to rule out the switches themselves.

Let me know if you think of anything else I can try.  Is the double action what's called ghosting?

TD22057, 2016-06-09:
Try moving swapping the pins that col 15 and 16 are connected to with col 1 and 2 (or some others).  If they behave the same (badly), then you know it's the wiring.  If col 1 and 2 start behaving badly, then you know it's the software.  Use the debug flag in matrix.cpp, that's closer to the source than the magic.

jasonm23, 2016-06-10:
I decided to rewire to different teensy pins also.

I'll let you know how it goes, 8hrs + from now

tmk, 2016-06-10:
Probably this.
https://github.com/tmk/tmk_keyboard/wiki/FAQ#cant-read-column-of-matrix-beyond-16

Check following parts.
https://github.com/TD22057/tmk_keyboard/blob/shark/keyboard/shark/matrix.cpp#L71
https://github.com/TD22057/tmk_keyboard/blob/shark/keyboard/shark/TeensyPin.hpp#L21-L27

jasonm23, 2016-06-10:
Thanks Hasu @tmk, saved me a bout of soldering, head scratching and face palming.

TD22057, 2016-06-10:
Good catch @tmk.  I think TeensyPin is fine - the registers are all 8 bits.  I think only that matrix row needs to change.  This shows the down side of matrix_row_t.  Since it's auto-typedef'ed based on the column size, it makes it difficult to look at the code and see that it's incorrect.  Forcing people to use the right variable sizes is more work, but it's also easier to understand and find problems.

closed 2016-06-10 by tmk

jasonm23, 2016-06-10:
Please reopen, I have yet to fix this, just have ideas to go on.

jasonm23, 2016-06-10:
fixed:

```
// shark/matrix.cpp
bool matrix_is_on(uint8_t row, uint8_t col)
{
    return (s_matrix[row] & ((matrix_row_t)1UL<<col));
}
```

```
// shark/TeensyPin.cpp

// Set the bit number 'bit' in the variable 'field' to 1.
#define BIT_SET( field, bit ) (field) |= ( 1UL << bit )

// Set the bit number 'bit' in the variable 'field' to 0.
#define BIT_CLEAR( field, bit ) (field) &= ~( 1UL << bit )

// Return the bit number 'bin' from the variable 'field'.
#define BIT_READ( field, bit ) (field) & ( 1UL << bit )
```

jasonm23, 2016-06-10:
Thanks again @tmk Hasu