tmk/tmk_keyboard #357: hook_layer_change has the wrong value after setting ACTION_DEFAULT_LAYER_SET

squarefrog, 2016-06-07:
There appears to be an issue with the new layer hooks.

Here I push my first layer, using `[1] = ACTION_DEFAULT_LAYER_SET(1),` and I get the following debug output.

```
Default layer set: 00000000000000000000000000000000
Default layer set: 00000000000000000000000000000010
```

Next, I toggle layer 3, using `[12] = ACTION_LAYER_MOMENTARY(2),`

```
Layer changed: 00000000000000000000000000000100
```

All good so far. Now I release the momentary button, and it states the layer is now 0:

```
Layer changed: 00000000000000000000000000000000
```

However this has actually popped back to layer 1 (which is intended behaviour). Functionally, I have no problem with this, but it does break my LED function:

```
void hook_layer_change(uint32_t layer_state)
{
  show_layer_led(layer_state);
  xprintf("Layer changed: %032lb\n", layer_state);
}

void hook_default_layer_change(uint32_t default_layer_state)
{
  show_layer_led(default_layer_state);
  xprintf("Default layer set: %032lb\n", default_layer_state);
}

void show_layer_led(uint32_t layer)
{
  /*
   * Set all LEDs off here.
   */
  switch (layer) {
      case 1 << 1:
          /* Set LED 1 on */
          break;
      case 1 << 2:
          /* Set LED 2 on */
          break;
      case 1 << 3:
          /* Set LED 3 on */
          break;
      case 1 << 4:
          /* Set LED 1 on */
          /* Set LED 2 on */
          break;
      default:
          /* do nothing */
          break;
  }
}
```

As you can tell this resets the LED for layer 1. Is there any smart work around?

tmk, 2016-06-07:
You can know actual current layer state with `(default_layer_state | layer_state)` from action_layer.h.

So call like `show_layer_led(default_layer_state | layer_state);` from those two hooks.

squarefrog, 2016-06-07:
Oh I like that 👍  thank you!

closed 2016-06-07 by squarefrog