tmk/tmk_keyboard #360: ADB Converter does not show up on USB bus

jceaser, 2016-06-12:
New here, not sure where to start, but I cloned a fresh copy of the code, compiled and installed, but no luck.

# The Teensy2.0 board does not show up on the USB bus unless I press the flash button
# No lite on board
# Wired board to keyboard using VCC, Ground, and PD0 with a 4.7k Ohm resistor on VCC and PD0.

Reasons I think the problem is here:
# http://guide-images.ifixit.net/igi/gAnVPQBeHPfVsRWX.large - it looks for all the world as if the blue wire is on PF0. Which is the data pin?
# https://github.com/tmk/blargg-adb-teensy worked "out of the box", sure I had to add "int" and "const" to a few lines, but it worked, so I know my wiring is good and my board is working.

So I could just move on and use blargg-adb-teensy but your project looks way more active and feature rich. What can I do to debug this thing? The code hear is a bit overwhelming, I would like to start with making the light blink and proving that the code is in fact loading and booting, unless someone here has a better suggestion.

jceaser, 2016-06-12:
I fixed my own problem but I'm unsure if I did it the right way, I changed the processor in the Makefile as shown below with a diff:

```
 # atmega32u4   Teensy2.0
 # atemga32u4   TMK Converter rev.1
 # atemga32u2   TMK Converter rev.2
-MCU = atmega32u2
+MCU = atmega32u4
```

1. is atemga32u4 spelled correctly?
2. should I have to change MCU or should the system figure it out?

tmk, 2016-06-25:
As the comment say you should use proper value for your hardware.

closed 2016-06-25 by tmk