tmk/tmk_keyboard #364: Arduino compatibility

tmk, 2016-06-22:
https://learn.adafruit.com/convert-your-model-m-keyboard-to-bluetooth-with-bluefruit-ez-key-hid/code-1
https://github.com/bgould/arduino_tmk_keyboard

bgould, 2016-06-22:
FYI - the code in my repo is based on a version of your firmware from a couple years ago... it is probably much easier to integrate with the Arduino libraries now.

Also back then I had to do some ugly things to make it so that you could build projects in the Arduino IDE... if I remember correctly there were problems such as not being able to pass flags to GCC and so couldn't change the version of the C spec that the compiler would allow.

Also, I didn't comment this very well but these two files are very important, it is necessary to include them in the Arduino project itself so that the library gets compiled properly in the IDE:

https://github.com/bgould/arduino_tmk_keyboard/blob/master/examples/terminal_bluefruit_converter/include_tmk.c
https://github.com/bgould/arduino_tmk_keyboard/blob/master/examples/terminal_bluefruit_converter/include_api.cpp

I can't remember exactly why that is necessary but I do remember that I got the idea from here:

https://github.com/adafruit/Adafruit-Trinket-USB/blob/master/TrinketKeyboard/usbdrv_includer.c

If you have any questions or need any help with this let me know and I'll try to lend a hand if I can

tmk, 2016-06-22:
Thank you for the info!