tmk/tmk_keyboard #370: HID Consumer page usages

tmk, 2016-07-18:
Some of HID Conumer page(0x0c) uasges are defined in `common/report.h`. Is there other usages used in OS(Mac/Linux/Windows) to be added?
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/report.h#L37-L72

HID Usage Tables pp75-97
http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

Let me know usages to be added.

cubiq, 2016-07-18:
The following would be the bare minimum and they should work on all OSes

0E2 Mute
0e9 volume increment
0ea volume decrement
0CD play/pause
0B0 play
0B7 stop
0b1 pause
0b3 fast forward
0b4 rewind
0b5 next track
0b6 previous track
0B8 Eject
192 Calculator
196 Internet Browser
1B4 File browser
06F brightness up (not sure about this) 
070 brightness down (not sure about this one either)
081 Power
082 Sleep
083 Wakeup (this never ever worked for me!)

Nice to have:

18B Email reader
1A0 Command Line Processor / Run
1C6 Research/Search Browser
1C7 Audio player
0C0 Frame forward
0C1 Frame back

yurikhan, 2017-01-23:
A while back I did a research on which Consumer page usages are usable with X11/Linux. I was not very interested in the functions assigned to them by default — mostly whether they pass through all abstraction layers and can be detected with XKB.

* Many usages are ignored in the `usbhid` kernel module and never make it to the userland.
* Some produce a Linux key code that is greater than 0xFF, which makes them unusable in XKB because XKB key codes are 8-bit.
* Some Consumer page usages are equivalent to Keyboard page usage (e.g. 0x0C 0x42 Menu Up translates to the same key code as 0x07 0x52 Up Arrow).
* Some Consumer usages are equivalent to other Consumer usages (e.g. 0x0C 0x82 AL Programmable Button Configuration and 0x0C 0x22A AC Bookmarks both translate into `KEY_BOOKMARKS`).

Here’s the link: <https://goo.gl/xigUiw>

All in all, there are 44 Consumer usages that are usable and distinguishable from all other usages.

tmk, 2017-01-25:
Thanks for the info. I'll refer these when refactoring the code.

dctucker, 2017-02-04:
Just wanted to comment that I was able to make the brightness up and down keys work on Mac OS by using the 006F and 0070 key codes as mentioned above. This is pretty sweet since it means I don't need to use Karabiner or customize my keyboard shortcuts using System Preferences to use these media/consumer functions.

eltang, 2017-02-04:
@dctucker Why not just use F14 and F15?

tmk, 2017-02-04:
@dctucker ,
You can send the codes even if they are not defined for in TMK. Just use `ACTION_USAGE_CONSUMER(id)`.

Let us know whether those codes works or not in fact.

dctucker, 2017-02-05:
They do in fact work. The reason I don't want to use F14 and F15 is I have other key-mappings for them (switch to desktop left/right), and one of the design goals of the keyboard I'm building is to replace my old Apple Keyboard with Numpad which has Fn+F1/F2 bound to brightness down/up.

tmk, 2017-02-05:
Great. Thanks.
006F and 0070 is marked as "reserved" in HID usage table spec, probably Apple specific usages.

tmk, 2017-02-11:
Display Brightness Controls:
http://www.usb.org/developers/hidpage/HUTRR41.pdf

I confirmed 0x6F and  0x70 works also on Windows 10.
```
ACTION_USAGE_CONSUMER(0x6F)
ACTION_USAGE_CONSUMER(0x70)
```

tmk, 2017-02-11:
And just confirmed Brightness Increment and Decrement(0x6f and 0x70) work with Ubuntu16.04 on Thinkpad x201s too.

tmk, 2017-06-13:
`AL Email Reader(18A)` is recognized as 'Photo viewer' on Windows10 weirdly.

https://geekhack.org/index.php?topic=88720.msg2439662#msg2439662

tmk, 2019-01-27:
Define `AC_` name for your action like this line in your unimap.c before using it,
```
#define AC_xxxx ACTION_USAGE_CONSUMER(0x6F)
```
And then place `xxxx` in your `UNIMAP()`.

tmk, 2019-01-27:
Reopen this.

tmk, 2020-05-11:
Added support for Display Brightness Control at 15434c3a167f0c992e0f55af5d1d6d2ea6a31878

https://github.com/tmk/tmk_keyboard/blob/15434c3a167f0c992e0f55af5d1d6d2ea6a31878/tmk_core/common/actionmap.h#L422-L423

https://github.com/tmk/tmk_keyboard/blob/15434c3a167f0c992e0f55af5d1d6d2ea6a31878/tmk_core/common/keycode.h#L177-L178