tmk/tmk_keyboard #375: Is there any method to keep the default layer even keyboard is disconnected?

chenzhihuai1990, 2016-08-23:
I have use some key to change the default layer, but when I disconnect the keyboard and reconnect it, the default layer is restored to old one. How could I store the default layer state when keyboard is ejected?

tmk, 2016-08-23:
You can use boot magic, maybe?
https://github.com/tmk/tmk_keyboard#default-layer

kaihuang201, 2017-10-31:
You can add one line of code to make default layer setting persistent across power cycles.
https://github.com/kaihuang201/tmk_keyboard/commit/75d4358f2d2a106c220bb8eea7ebdd68e647ee43
This writes the default layer to eeconfig everytime it is changed.

I will create a pull request for this, probably also make it configurable with a compile flag.