tmk/tmk_keyboard #376: Instructions for building and instilling on OS X

wmalloc, 2016-08-23:
Would be possible to add a wiki page on how to do this?

tmk, 2016-08-23:
Yeah, it would be appreciated. Feel free to add a page for this!
I think you can use macports or homebrew basically and refer to qmk doc.
https://github.com/jackhumbert/qmk_firmware/blob/master/doc/BUILD_GUIDE.md#mac

tmk, 2016-08-23:
Also crosspack is available.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/doc/build.md#1-install-tools
https://www.obdev.at/products/crosspack/index.html

Only real OSX users know which method is preferable and write detail instructions. Contributions are welcomed.

matteomatassoni, 2016-08-26:
I was able to do this following the instruction from http://maxembedded.com/2015/06/setting-up-avr-gcc-toolchain-on-linux-and-mac-os-x/

pyro2927, 2017-01-24:
Just built http://github.com/tmk/tmk_core on my MacBook Pro running macOS 10.12 and can confirm that all I needed to do was @wmalloc's instructions:

```
brew tap osx-cross/avr
brew install avr-libc
brew install dfu-programmer
```

tmk, 2017-01-25:
Just added an entry for this in wiki. Thanks guys!
https://github.com/tmk/tmk_keyboard/wiki#build-on-mac