tmk/tmk_keyboard #378: internal compiler error

spirosag, 2016-09-07:
I am trying to compile for a s60-x an already written layout but getting the following error:

-------- begin --------
avr-gcc.exe (WinAVR 20100110) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mkdir -p obj_S60-X_lufa/common
Compiling C: ../../tmk_core/common/keymap.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=4096 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DVERSION=600bc75 -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_S60-X_lufa/common/keymap.lst -I. -I../../tmk_core -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_S60-X_lufa_common_keymap.o.d  ../../tmk_core/common/keymap.c -o obj_S60-X_lufa/common/keymap.o
../../tmk_core/common/keymap.c: In function 'keymap_fn_to_action':
../../tmk_core/common/keymap.c:246: internal compiler error: in iterative_hash_expr, at tree.c:5363
Please submit a full bug report,
with preprocessed source if appropriate.
See <URL:http://sourceforge.net/tracker/?atid=520074&group_id=68108&func=browse> for instructions.
make: *** [../../tmk_core/rules.mk:553: obj_S60-X_lufa/common/keymap.o] Error 1

Any ideas? Because I don't know how to debug the internal compiler error

shayneholmes, 2016-09-07:
Just came here to post this, pulled the latest master branch into my repo and am getting the same error. I'm on Windows 10.

> -------- begin --------
> avr-gcc (WinAVR 20100110) 4.3.3
> Copyright (C) 2008 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> mkdir -p obj_ergodox_lufa/common
> Compiling C: ../../tmk_core/common/keymap.c
> avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=512 -DFLASH_SIZE_BYTES=0x8000 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DNKRO_ENABLE -DSLEEP_LED_ENABLE -DNO_SUSPEND_POWER_DOWN -DVERSION=unknown -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_ergodox_lufa/common/keymap.lst -I. -I../../tmk_core -I../../tmk_core/protocol -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_ergodox_lufa_common_keymap.o.d  ../../tmk_core/common/keymap.c -o obj_ergodox_lufa/common/keymap.o
> ../../tmk_core/common/keymap.c: In function 'keymap_fn_to_action':
> ../../tmk_core/common/keymap.c:246: internal compiler error: in iterative_hash_expr, at tree.c:5363
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <URL:http://sourceforge.net/tracker/?atid=520074&group_id=68108&func=browse> for instructions.
> make: *** [obj_ergodox_lufa/common/keymap.o] Error 1
>

tmk, 2016-09-07:
WINAVR is too old now and not much compatible to modern Windows. Try to use XP if you want to stick on WinAVR.

spirosag, 2016-09-07:
Is there an alternative on Windows?

tmk, 2016-09-07:
Why do you guys use WinAVR? I mean, where did you get the idea?
I'll fix if TMK docs says that somewhere wrongly.

shayneholmes, 2016-09-07:
Reverting 19dca3def8560888c0492072511ff90f8ad17df6 fixed this for me, is compiling fine without that commit.

tmk, 2016-09-07:
@spirosag sure, Linux and OSX have no problem, I believe.

shayneholmes, 2016-09-07:
I'm using WinAVR because it was the guidance when I set out to program my new Ergodox three years ago (!), and it has been ticking merrily along since. I set up my keymap on it back then, and since then I occasionally tweak and recompile.

spirosag, 2016-09-07:
@tmk forgive me it is the first time I am compiling and have read that winavr is the way. Since it is deprecated is there another way, or you can't compile on Windows?

tmk, 2016-09-07:
@shayneholmes haha, 3 years :) glad to hear you found solution.

tmk, 2016-09-07:
@spirosag check README and build doc, first. And wiki pages possibly.
https://github.com/tmk/tmk_keyboard#build-firmware-and-program-controller

If you don't mind intalling huge VM and image this is easy way for Windows users.
https://github.com/tmk/tmk_keyboard/wiki/Build-firmware-on-VirtualBox

spirosag, 2016-09-07:
@tmk oh, I did that's why I asked! When I said windows I meant without vm etc. Thanks anyway for the proposal. Should we wait for a fix that will restore the compile error with winavr now that @shayneholmes found it? In any case thanks a lot for your help!

tmk, 2016-09-07:
WinAVR had been the best toolset and very easy to install, but it is too old, has not been maintained long time and has some problems with modern Windows and TMK. TMK doesn't support build with WinAVR anymore. Fix for this won't appy to TMK unless TMK codes is clearly incorrect.

You still may be able to compile with WinAVR applying some fixes. But it is not recommended. Use new toolset.

spirosag, 2016-09-08:
@tmk just one last question. In the wiki it is mentioned to "install Atmel AVR Toolchain for AVR GCC compiler and Cygwin(or MinGW) for shell terminal". Is this used in order to compile? I have both cygwin and atmel avr toolchain but having a little trouble. Have I understood something wrong?

tmk, 2016-09-08:
@spirosag I didn't mean you can't compile on windows, you just need time to learn it for support from someone with skill. It is shame that you could not get proper support from manufacturer or seller.

spirosag, 2016-09-08:
@tmk oh, absolutely! thanks for the quick replies and all your support

closed 2016-09-08 by spirosag

tentator, 2016-11-18:
( FYI a fix to this issue can be a change this line in [...]/tmk_core/common/keymap.c:246: )
return (action_t){ .code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]) };