tmk/tmk_keyboard #381: Fix OS X Recognizing keyboard as Mouse/Tablet

tmk, 2016-09-08:
interesting.

> Modified the LUFA USB HID Descriptor to change the logical/usage
minimums for System Control from 0x01 (Mouse) to 0x81 (System Power
Down), this fixes OS X recognizing the Planck as having a mouse and
tablet, even with mousekeys off.

https://github.com/jackhumbert/qmk_firmware/commit/d48a4cf1eb80600fb3e8eb859e5b77c954a18137

tmk, 2016-09-08:
maybe, #74 is related to this

exiva, 2016-09-21:
I don't have the issue referenced in #74 with the fix I did at https://github.com/jackhumbert/qmk_firmware/commit/d48a4cf1eb80600fb3e8eb859e5b77c954a18137 on a Planck.

tmk, 2016-09-21:
@exiva Thanks!
I did cherry-pick the commit at b2d791804d some days ago.

closed 2016-09-21 by tmk