tmk/tmk_keyboard #382: The naming of cats ... errr ... keymap.c

DreymaR, 2016-09-09:
Just a tiny issue:

Could the ***keymap.c*** in converter/usb_usb be named ***keymap_all.c*** instead?

• It's logical since it does in fact contain KEYMAP_ALL definitions
• It'd be more practical for me since I use a script that calls keymaps as 'keymap_<name>.c'
• I don't think it'd affect anything apart from line 129 in Makefile (that sets default keymap)

From: **SRC := $(KEYMAP_FILE).c $(SRC)**
To: ** SRC := keymap_all.c $(SRC)**

I don't think unimap or actionmap are the default values anyway? But I may be wrong. :-)

tmk, 2016-09-09:
maybe, keymap.c should use KEYMAP() instead of KEYMAP_ALL()? I might want to rename KEYMAP() to KEYMAP_ANSI().

anyway, I'll reorganize keymap files if I have time.

DreymaR, 2016-09-09:
I don't know where you want to go with the new UNIMAP. But I feel there may be too many keymaps now?

The way I see it, it might make sense to have one simple type that supports ANSI+ISO (because they're really similar), probably one KEYMAP_JIS (or it could just use ALL) and one KEYMAP_ALL or UNIMAP.

As it is now, there are some layouts to choose from in the keymap(_all).c file, but for instance the keymap_iso.c file is just proof-of-concept. It's too much work to maintain and write layouts for many keymap files, so it'd be nice to not have too many!

tmk, 2016-09-10:
Cleaned up keymapfiles.
https://github.com/tmk/tmk_keyboard/commit/e957c1c6f1c36981ca7823396565344e99b06690

I might want to remove KEYMAP_(ISO|JIS|ANSI) now for simplicity. Merith of using them seems to be very small, you can always use KEYMAP_ALL(KYEMAP now).

tmk, 2016-09-10:
Oops, that commit broke existent keymaps.

Revert macro names for backward compatibility.
https://github.com/tmk/tmk_keyboard/commit/659ae2375e66fd140319aeba9540061d2fb3aa28

DreymaR, 2016-09-10:
> 659ae23

Ooo, I'd strongly prefer that all keymaps had a name! So you should use
KEYMAP_ALL and not just KEYMAP. That'd unbreak things, too, right? :-)

tmk, 2016-09-10:
I reverted its name to `KEYMAP_ALL()`, macro names didn't change in the end :D

closed 2016-09-10 by tmk