tmk/tmk_keyboard #383: Kernel warning messages

sadboy, 2016-09-16:
Every time I plug in my HHKB I get these three warning messages from my kernel

    usb 3-2.4.4.1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
    usb 3-2.4.4.1: ep 0x82 - rounding interval to 64 microframes, ep desc says 80 microframes
    usb 3-2.4.4.1: ep 0x83 - rounding interval to 64 microframes, ep desc says 80 microframes

But otherwise keyboard functions perfectly. This is version 'v2.9' on Ubuntu 4.4.0-36-generic kernel.

tmk, 2016-09-16:
Is that really warning? I think it is just info and no problem.

Are you using Intel chip set? I guess this is behaviour specific to Intel USB host controller.
Use AMD if you don't want to see those messages :) Ive never seen them so far.

sadboy, 2016-09-16:
> Is that really warning?

Yes, it's in 'dmesg -l warn'.

> Are you using Intel chip set? I guess this is behaviour specific to Intel USB host controller.
> Use AMD if you don't want to see those messages :) Ive never seen them so far.

Interesting. These are the two controllers I use, and they both generate the same warnings:

    USB controller: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31)
    USB controller: Fresco Logic FL1100 USB 3.0 Host Controller (rev 10)

tmk, 2016-09-17:
I don't know why this is warn but you can ignore.

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1491657

closed 2016-09-17 by tmk