tmk/tmk_keyboard #384: Sun Type 5c  'Diamond' keys dead

bjd-pfq, 2016-09-19:
On a Sun Type 5c keyboard, I was using a tmk_keyboard version from January 2015.
That version happily recognized the two 'diamond' keys immediately left and right of the space bar.
I'm using Linux, and in xev these keys are reported as Hyper_L (keycode 133) and Hyper_R (keycode 134).

Now I am trying the current tmk_keyboard (on a second Sun Type 5c and a second Teensy 2.0), and this version leaves the two diamond keys stone cold dead. The 'old' teensy, when switched to this other keyboard, works fine, so the glitch must be in the software.  Same system, no reboots (in that regard, testing is a cinch).

Any hints?

tmk, 2016-09-19:
What did you change source something?

bjd-pfq, 2016-09-19:
No, I didn't change anything. But the one keyboard runs on tmk_keyboard from early 2015 (I think before the refactoring of the entire tree), and the other one I am testing runs on the current version of tmk_keyboard. 

Bothe keyboards are attached to the same system.  They work fine, only the newer version has the two dead keys.

In fact, I had mapped my MAGIC KEY in the old version to the combination of the l]Left and Right diamond keys. Works like a charm, but I cannot replicate this even in the sources for the current version, for I don't know and don't recall what I used for the symbols in the IS_COMMAND #define :-/ for the two diamond keys.  I perused the code today, but I cannot determine what they are.

tmk, 2016-09-19:
Did you change Makefile to use Teensy? Now Makefile is configured for TMK converter with ATMega32u2 by default.

bjd-pfq, 2016-09-19:
Yes, all the settings have been checked and double checked 36 times and they all are correct.

Specifically, I have, in config.h:

/*#define USE_LEGACY_KEYMAP*/

and in Makefile:

MCU = atmega32u4       # Teensy 2.0

MOUSEKEY_ENABLE = yes	# Mouse keys
EXTRAKEY_ENABLE = yes	# Audio control and System control
CONSOLE_ENABLE = yes	# Console for debug
COMMAND_ENABLE = yes    # Commands for debug and configuration
#NKRO_ENABLE = yes	# USB Nkey Rollove

But do you know what key symbols I would need for these two Diamond keys in the IS_COMMAND C-macro #define??  I cannot find any good candidates anymore in keycodes.h, but I must be overlooking some symbols.

bjd-pfq, 2016-09-19:
Sorry, the USE_LEGACY_KEYMAP is undefined in config.h.  But it doesn't matter, I tried with and without, and it doesn't bring these keys alive in xev.

tmk, 2016-09-19:
I just confirmed the latest code still works for me with Topre Sun keyboard.

tmk, 2016-09-19:
yes, you can cofigure magic key with IS_COMMAND macro.

bjd-pfq, 2016-09-19:
But which symbols do I use?  I thought it was KC_HENK and KC_MHEN but that didn't work.

Also, I think xev should at least give /some/ output, whether or not I have programmed these keys for the teensy, should it not?

tmk, 2016-09-19:
Try getting plain the latest source code then just change MCU in  Makefile.
And `make clean` and `make` to  get firmware.

if it doesn't help, you might want to check your toolchain. I use gcc4.9.2, btw.
```
$ avr-gcc --version
avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tmk, 2016-09-19:
Sun diamond key is equivalent to Windows key or Mac Command key, which are called Gui key in TMK and USB spec. See around p59.
http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

bjd-pfq, 2016-09-19:
LGUI and RGUI, you mean.  These are defined already by default in the IS_COMMAND !
Still it doesn't work.
Let me just try a fresh new tree, like you suggest.
...
<time passes>

bjd-pfq, 2016-09-20:
No go.  Just compiled and installed a fresh new tree, and the keys are just dead, whereas they work under the old style teensy (in fact I have both keyboards hanging off my current system :-)

The only thing I changed in all the source is MCU = atmega32u4.

Btw., I too have avr-gcc (GCC) 4.9.2.

bjd-pfq, 2016-09-20:
Btw., I tried to recompiled,, this time having commented out USE_LEGACY_KEYMAP, and the compile
aborts with:

/home/bjd/Documents/hw/kbd/sunkbd/tmk/tmk_keyboard-master-20160918-orig/converter/sun_usb/../../tmk_core/common/keymap.c:246: undefined reference to `fn_actions'
/home/bjd/Documents/hw/kbd/sunkbd/tmk/tmk_keyboard-master-20160918-orig/converter/sun_usb/../../tmk_core/common/keymap.c:246: undefined reference to `fn_actions'
collect2: error: ld returned 1 exit status
../../tmk_core/rules.mk:550: recipe for target 'sun_usb.elf' failed
make: *** [sun_usb.elf] Error 1

omenlabs, 2016-09-20:
I use a Sun Type 5c daily and make heavy use of the diamond (meta) keys.  It has been a while since I updated, but I can give it a shot.

bjd-pfq, 2016-09-20:
Caveat: better use a second teensy; I'm glad I did...

bjd-pfq, 2016-09-20:
What baffles me is that to begin with, the keys should generate at least some response from xev.
The 'old style tmk' keyboard generates this for the left and right diamond keys:

KeyPress event, serial 37, synthetic NO, window 0x5000001,
    root 0x294, subw 0x5000002, time 396448473, (47,32), root:(1229,663),
    state 0x0, keycode 133 (keysym 0xffed, Hyper_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 37, synthetic NO, window 0x5000001,
    root 0x294, subw 0x5000002, time 396448561, (47,32), root:(1229,663),
    state 0x40, keycode 133 (keysym 0xffed, Hyper_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

tmk, 2016-09-20:
Gui, Diamond, Windows and Command keys are usually recognized as Super in Xorg on my site. I don't know about Hyper and Xorg key system much, though.

omenlabs, 2016-09-20:
I just cloned and flashed and my diamond keys are working.  I'm using an m32u4 (arduino micro) and hardware serial.

tmk, 2016-09-20:
@omenlabs Great.
BTW, I use software serial without pull-up resistors or any extra component.

tmk, 2016-09-20:
@bjd-pfq try this hex file which is compiled with software serial for atmega32u4, it should work with teensy.
https://drive.google.com/file/d/0Bygvtwt-Y9HhOXgwOUVITThIc2s/view?usp=sharing

omenlabs, 2016-09-20:
@tmk The timing was off enough with the software serial that my Sun Type 5c wouldn't recognize commands (according to my scope it was a small ~850us vs expected 833us per symbol).  I did the PR for hardware serial support a while back and have been using the type 5c at work ever since.

tmk, 2016-09-20:
@bjd-pfq If you are using Xmodmap or XKB to tweak keys I guess it might cause your problem.

bjd-pfq, 2016-09-20:
@tmk Thanks for the hex file.  Installed it, and ..... no go!
I suspect my teensy is flakey?
Everything works like a charm, all the FN keys on the left of the Sun kbd are recongized, the Help key is recognized (this is all exactly like my other compiles, btw.), but those Diamond keys remain stone cold dead.
Btw., these are two oriingal Sun Type 5c keyboards, partno. 3201234-02 for both; and both are hanigijng off of my Linux box at the moment (though I am constantly plugging off the default keyboard and power cycling the one under test, to see if that changes anything, but it doesn't).

tmk, 2016-09-20:
@omenlabs Interesting. I tried software serial with type 5 for short period and it worked. But I will have to check its signals some later, anyway I believe  I can fix or tweak code in the end beuase sun uses very slow 1200baud signal. Maybe it needs assembly coding which requries me to learn more, fun! :D

bjd-pfq, 2016-09-20:
No. Checked that earlier,.xmodmap is empty. Donb't call xmodmap any other way. Don't have xbindkeys or the like. 

And btw., shouldn't that affect /both/ keyboards I'm having attached to my Linux box?

omenlabs, 2016-09-20:
I have the same part number.  Have you tried the HID listen program and debug mode?

bjd-pfq, 2016-09-20:
The teensy runs the blink test program fine, btw., but I suspect that doesn't do much of stress testing...

bjd-pfq, 2016-09-20:
@omenlabs Good point.  An hour ago, before this flurry of posts started, I was just running hid_listen (as root), and it produced the initial messages,, but then I couldn't get it into debug mode in any way.
Let me pick up on that..., thanks for the reminder.

bjd-pfq, 2016-09-20:
Update:

I hooked up the 'old-style-tmk' teensy to the Sun keyboard with the dead Diamond keys.  What do you know -- still these keys are dead!

Which leads me to the conclusion there is someting amiss in this keyboard.

To verify, I hooked up the 'old-style-tmk- teensy to a /third/ Sun keyboard (same partno.).
An lo and behold, there the Diamond keys are responding correctly.

In short: the second teensy is prob. not flakey, but the second Sun keyboard itself has a flaw!
Next thing is to dissemble it, and look at the PCB and the conductive tracing for these two keys.

bjd-pfq, 2016-09-20:
Update 2:

Lane 2 on the large membrane connector on the PCB (connecting only the two Diamond keys!) was not making contact.
I wasn't able to fix it either, so what I did was swap the entire 3-layer membrane stack + PCB from the 3rd Sun Type 5c keyboard (which was a partly cannibalized keyboard anyway) to the keyboard that was faulty.

Problem solved.

omenlabs, 2016-09-20:
Glad you got your keyboard working again!  :-)

bjd-pfq, 2016-09-20:
Thanks, and thanks to you and tmk for your input.

Might I add a late final question: do your ScrollLock and Compose-key LEDs light up when these keys are pressed?  Mine stay off, and only come on when the keyboards are plugged in, when they are reset, or when the MAGIC+all LEDs on is pressed.

omenlabs, 2016-09-20:
I see the same issue with the lights not coming on.

bjd-pfq, 2016-09-20:
Thank, I am consoled.... (double pun intended).
Maybe it should be reported as an issue?

tmk, 2016-09-20:
I don't think Xorg/Linux usually sends signal for ScrollLock and Compose to keyboard, my desktop doesn't send for ScrollLock at least.(I don't have compose key) Try Windows 10, it seems to send for ScrollLock. Or you can xset to send the signal to keyboard.(maybe xset led on?)

If your OS support those indicators it should work. See led.c and send PR open new issue if not.

closed 2016-09-20 by tmk

bjd-pfq, 2016-09-20:
Ah, but both /are/ recognized by xev, so I can use them, or reprogram them (with xmodmap), etc.;
here's the xev output:

ScrollLock:
KeyRelease event, serial 37, synthetic NO, window 0x3a00001,
    root 0x294, subw 0x3a00002, time 31302858, (49,53), root:(1231,684),
    state 0x0, keycode 78 (keysym 0xff14, Scroll_Lock), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

Compose:
KeyPress event, serial 37, synthetic NO, window 0x3a00001,
    root 0x294, subw 0x3a00002, time 31308714, (49,53), root:(1231,684),
    state 0x0, keycode 135 (keysym 0xff67, Menu), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

tmk, 2016-09-20:
First, indicator status is generally controlled by OS/host, not keyboard(or converter). If your OS doesn't controll them LEDs doesn't lit. And Linux/Xorg doesn't seem to controll indicators for those keys. Again try WIndows or xset.

Can you compose charactor with your "compose" key? I tried to setup compose key on xfce4/Ubuntu16.04 and xev recognizes it as 'Multi_key', not 'Menu' and I can compose now. But Xorg still doesn't handle indicator of compose.

I think indicators of compose and scrollLock are lit on with 'xset led on', did you try it?

bjd-pfq, 2016-09-20:
Yes, that works, although oddly enough not ever for CapsLock or NumLock.
and only numerically for Compose (reported by xev as Menu):

xset led 3		sets ScrollLock    works
xset led 5		sets Compose      works

xset led named "Scroll Lock"    works
xset led named "Menu"               xset: Invalid led name: Menu

I suppose you can program Compose to use as a compose-key; but I have been using the one right next to it, AltGraph, for that purpose some time.   Strangely, since you mentioned 'Multi-key', my AltGraph-key is reported by xev as 'Multi-key' (which indeed more or less is synonym with 'compose').

bjd-pfq, 2016-09-21:
In Linux Mint, which I am using, I have not done anything special for the keyboard, not in the cinnamon-settings, not in xmodmap or any other way.  But I have a feeling that in this default setup there are some misnomers.  The Compose key (with the word Compose on it, and it has a led) should more appropriately be called 'Multi-key'.  And AltGraph?  I don't know.

I do know there is an option in the keyboard settings 'Maintain Key compatibility with old Solaris keycodes'.
I haven't turned that on yet, but I am going to try it next, to see what difference if any that makes.

tmk, 2016-09-21:
Great, the converter supports all indicators! No issue there.

Yes,  I just read this and configure my compose(Menu) key as 'Compose key(Multi key)' on Xorg.
https://help.ubuntu.com/community/ComposeKey

How those indicators are handled depends fully on your host(OS), you may configure them or not.
I never used actually Compose key and AltGr before and don't know much. You will get more help from Mint community.