tmk/tmk_keyboard #386: Problem on Sleep/Wake/Wakeup from keyboard

tmk, 2016-09-23:
It sometime works but sometime not. Iffy.

tmk, 2016-09-27:
After test with HHKB controller with ATmega32u4 it doesn't seem to have problem.

tmk, 2016-10-05:
PS/2 converter with ps2_interrupt.c(PS2_USE_INT) had problem on handling signal when suspended. In suspend state microcontroller falls into power down mode for 15ms until it checks keyboard for wakeup condition(keypress by user). During the 15ms sleep it can miss signals or get wrong scan codes from keyboard and internal states of converter are corrupted.

WORKAROUND: Instead of power down, PS/2 converter uses standby mode now with `SUSPEND_MODE_STANDBY `. In standby mode it can wake by pin change interrupt and no longer miss signal from keyboard.

With ps2_busywait.c converter inhibits to send signal until it is ready to handle, so the problem above won't occur. ADB and M0110 also won't suffer from this probably because they send codes only when requested.(not confrimed)

With ps2_usart.c converter will have the problem. To solve it the converter will have to inhibit keyboard to send during microcontroller sleeps.

ANOTHER WORKAROUND:
If power saving is not important you can keep microcontroller running even when suspended so that converter shouldn't miss signals. Keyboard connected to converter won't sleep basically and vintage keyboards keep drawing much current all the time. No power saving of microcontroller is actually useless there.

Elizafox, 2019-02-20:
Issues with suspend/wakeup occur with terminal_usb. Issues occur with my Unicomp 3270 terminal keyboard.

tmk, 2019-02-20:
Probably you should make new issue and describe your problem.

This issue is mainly related to key responsiveness when you wake your computer with key press.

closed 2019-05-10 by tmk