tmk/tmk_keyboard #392: Suggestion: Better typing macros (hopefully easily implementable)!

DreymaR, 2016-09-30:
**TMK suggestion:**
The MACRO function is often used to type hotstrings. Could this be implemented more smoothly?

String handling as such may well be overmuch and inflexible, but much could be gained by some hopefully simple additions:
• A T() macro function that can take a key sequence as input instead of just one key.
    Then, instead of T(H), T(E), T(L), T(L), T(O), you can use T(H, E, L, L, O) which is a _lot_ nicer I think!
• A S() macro function that similarly shift-types one or several keys.
    That is, S(<stringseq>) equals D(LSHIFT), T(<stringseq>), U(LSHIFT)

To type 'Do NOT use!': S(D), T(O, SPC), S(N, O, T, SPC), T(U, S, E), S(1).
To type a smiley: S(SCLN), T(MINS), S(0).

Could this be added, please? It'd be fully backwards compatible.

tmk, 2016-10-03:
Thanks for the suggestion! i'll take into account when I redesign macros.

DreymaR, 2016-10-03:
While we're talking Macro redesign, one little thing: Is the **END** mark necessary? Naïvely I'd think that when the MACRO function reaches its last argument the macro is finished, and there's no need to end a macro at any other time than this.

If you could remove the need for END (just ignoring it if found), that'd be backwards compatible – unless someone actually found the need to end a macro before the last argument which I doubt.

DreymaR, 2016-10-21:
Rethinking the issue...:

What'd be even nicer than a S() function, would be syntax similar to AutoHotKey in the T() function. I don't know how hard that'd be to implement, but it should be like this:
**+/^/!/#** in front of key name means **Shift/Ctrl/Alt/Gui** are down (can be combined)

The S() function wouldn't be very practical since typing is usually in mixed case.

So now, my examples would be:
**Do NOT use!** – **T(+D, O, SPC, +N, +O, +T, SPC, U, S, E, +1)**
**:-)** – **T(+SCLN, MINS, +0)**
**DreymaR** then **Ctrl+Enter** (to send email) – **T(+D, R, E, Y, M, A, +R, ^ENT)**

DreymaR, 2017-01-16:
Another rethinking (given that the above suggestion may well be too hard to implement):

It'd be silly and clunky to have one **T()** and one **S()** function, after all. But could there simply be a few shortcuts for the **T()** function instead?
• SFNX for "next character (only) is shifted"
• SFDN/SFUP for "(left) shift down/up" (for longer shifted sequences)
• If you do the above, I suppose it'd be handy with DN/UP for ALt/CTrl/WIn as well. Maybe NX too?

Then:
**"Do NOT use!" – **T( SFNX, D, O, SPC, SFDN, N, O, T, SFUP, SPC, U, S, E, SFNX, 1)**
**":-)" – T( SFNX, SCLN, MINS, SFNX, 0)**
**"DreymaR" then Ctrl+Enter – T( SFNX, D, R, E, Y, M, A, SFNX, R, CTNX, ENT)**
**ALT+Tab – T(ALNX, TAB)**
**Ctrl+Alt+Del – T( CTNX, ALNX, DEL) ... if you can – if not, T(CTDN, ALNX, DEL, CTUP)**

Again, backwards compatible syntax. ;-)

DreymaR, 2017-11-13:
Any chance of something like this happening soon? 👍 

New thought: Could this be an easy-ish way of supporting glyph input? For Windows:
**"å" – T( ALDN, P0, P2, P2, P9 )**
**"Æ" – T( ALDN, P0, P1, P9, P8 )**
etc

tmk, 2017-11-14:
As for unicode input it would be useful to have dedicated method, like QMK or Kaleidoscope?
Kaleidoscope gives us clear idea on this.
https://github.com/keyboardio/Kaleidoscope-Unicode
https://github.com/keyboardio/Kaleidoscope-HostOS

DreymaR, 2017-11-23:
Maybe... but then again, if such a method is memory costly that's a consideration too! Unless we all migrate to the newest generation Teensy, memory is still limited.

So it might not be so bad to implement it per-platform, as a macro that may furthermore be aliased:
```#define A(a, b, c, d) (T( ALDN, Pa, Pb, Pc, Pd, ALUP ))```

Unless someone needs their device to support multiple platforms with this input, that'd be trickier. I'd make two keymap layers then, say, one for Windows and one for Linux, with different macros. But then again, that'd cost memory so... heh.

DreymaR, 2017-12-15:
As a quick solution for now, here's how shifted and Ctrl could be implemented easily (I think?):

In tmk_core/common/action_macro.h:
```
/* key type */
#define T(key)          TYPE(KC_##key)
/* key type with shift */
#define ST(key)          DOWN(KC_LSFT), TYPE(KC_##key), UP(KC_LSFT)
/* key type with ctrl */
#define CT(key)          DOWN(KC_LCTL), TYPE(KC_##key), UP(KC_LCTL)
```

DreymaR, 2018-01-24:
I have tested the above and it works fine. Could this or a variant of your choice be added to tmk_core/common/action_macro.h please?
```
/* Macro definitions for typing modified keys more easily in TMK macros                                               */
/*      - Example: Ctrl+Alt+[Shift+]<key> can be entered with CTL_( ALT_( [S]T(key) ))                                */
/*      - AGR_ is for RAlt/AltGr input, useful with international system layouts.                                     */
#define ST(key)    D(LSFT),   T(key),    U(LSFT)
#define SFT_(...)  D(LSFT), __VA_ARGS__, U(LSFT)
#define CTL_(...)  D(LCTL), __VA_ARGS__, U(LCTL)
#define ALT_(...)  D(LALT), __VA_ARGS__, U(LALT)
#define AGR_(...)  D(RALT), __VA_ARGS__, U(RALT)
/* Unicode macro for XOrg : #define UNI_(d1,d2,d3,d4)  CTL_( ST(U) ), T(d1), T(d2), T(d3), T(d4), T(ENT)              */
/* Unicode macro for MacOS: #define UNI_(d1,d2,d3,d4)  ALT_( T(d1), T(d2), T(d3), T(d4) )                             */
/* Unicode macro for Win  : #define UNI_(d1,d2,d3,d4)  ALT_( T(P##d1), T(P##d2), T(P##d3), T(P##d4) )                 */
#define UNI_(d1,d2,d3,d4)  ALT_( T(P##d1), T(P##d2), T(P##d3), T(P##d4) )
```

I don't know, you could make, say, UNIW_(), UNIM_() and UNIX_()  :-p instead of just the one if you like.

tmk, 2018-01-25:
NIce!
Could you make pull request for this?