tmk/tmk_keyboard #393: TrackPint support

tmk, 2016-10-06:
https://geekhack.org/index.php?topic=8971.msg2276591#msg2276591

https://github.com/Nephiel/tmk_keyboard/blob/master/tmk_core/protocol/ps2_mouse.c

tentator, 2016-11-07:
I tried Nephiels ps2 .c but it keeps my board in bootloader after flashing...

tmk, 2016-11-08:
Not sure what you mean, can you rephrase your problem and it's detail is needed.
And What s Nephiels?

tmk, 2016-11-08:
Opp s, ignore question about Nephiel. I didn't read first post.

tentator, 2016-11-08:
Hi Hasu,
 yes now what I'm trying to do is to sync with your latest upstream/master and see if then I get the new ps2_mouse.c of Nephiel working.. I'm rebasing it just now and actually I see some strange error while compiling my tentboard: 

`../../tmk_core/common/keymap.c: In function 'keymap_fn_to_action':
../../tmk_core/common/keymap.c:246: internal compiler error: in iterative_hash_expr, at tree.c:5363
Please submit a full bug report,
with preprocessed source if appropriate.
See <URL:http://sourceforge.net/tracker/?atid=520074&group_id=68108&func=browse> for instructions.
make: *** [obj_tentboard/common/keymap.o] Error 1
`

probably I still missed something.. did some fn_Actions maybe change their name in the meantime?
(like in https://github.com/tentator/tmk_keyboard/blob/master/keyboard/tentboard/keymap_tentboard.c)

Kr,
 tent:wq

tmk, 2016-11-09:
Your keymap file is old and not supported in current core, probaty. You need to revised it. See current keymap files located in projects under keyboard/ or converter/.

You mentioned bootloader in earlier comment, what is/was your problem actully?

tentator, 2016-11-09:
well a part that change towards "action_t" in the fn_actions I couldn't spot any other changes.. or where also keynames or action_function names changed? I can check out my file maybe -> https://github.com/tentator/tmk_keyboard/blob/master/keyboard/tentboard/keymap_tentboard.c ?  
Anyway what happened with the new ps2_mouse.c is that after flashing the teensy utility kept saying that it will disable it to prevent continuous reboots into bootloader and I tryed several compilations and trys but only going back to the older would make the keyboard come back to live.. thus now I was thinking to merge your latest changes into mine and check again...

tmk, 2016-11-10:
Yes, use `action_t` and other part of  your keymap looks fine. if you still get 'internal compiler error' try `make clean`. If it doesn't help I'd reccomend trying other versions of gcc.

tentator, 2016-11-10:
HI Hasu,
 yes I tried clean and with action_t but I think you are right when you say there's something strange with the compiler, I'm using avr-gcc and I'm under windoze and even if I compile something from your directory, latest fetch, I get errors, so it seems to be the gcc to be problematic.. but why only in recent versions? it was ok with older tmk from may or so..?

mkdir -p obj_hhkb/common
Compiling C: ../../tmk_core/common/keymap.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DBOOTLOADER_SIZE=4096 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DNKRO_ENABLE -DVERSION=2516993-dirty -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_hhkb/common/keymap.lst -I. -I../../tmk_core -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_hhkb_common_keymap.o.d  ../../tmk_core/common/keymap.c -o obj_hhkb/common/keymap.o
../../tmk_core/common/keymap.c: In function 'keymap_fn_to_action':
../../tmk_core/common/keymap.c:246: internal compiler error: in iterative_hash_expr, at tree.c:5363
Please submit a full bug report,
with preprocessed source if appropriate.
See <URL:http://sourceforge.net/tracker/?atid=520074&group_id=68108&func=browse> for instructions.
make: *** [obj_hhkb/common/keymap.o] Error 1

(this is by compiling your hhkb for instance)

what gcc would you suggest I could use? it needs to cross compile, right?

tentator, 2016-11-10:
I also tried to do compile in the "hbkb directory and do a make -f Makefile.lufa" and in this one I got a full compilation without errors..... uhm.....

tmk, 2016-11-11:
no idea :( File new issue for this if needed.  I don't think Trackpoint code causes your problem.

Old version of avr-gcc may cause problem like that, many people have had problem with WinAVR, in particular. Check this.
https://github.com/tmk/tmk_keyboard/wiki#build-firmware

tentator, 2016-11-11:
Well what do you mean with old version? I think the latest is WinAVR-20100110. Is there an alternative (for windows)? I have win7 . No virtualbox nor ubunto4windows (unless that works on win7) but have git-bash. .

tmk, 2016-11-11:
check this issue #378
TLDR; it is too old and have not been maintenanced for long time. Don't use WinAVR anymore.

tentator, 2016-11-11:
Hello Hasu,
 thanks for the hint but I think in the meantime I found the "bug" and with following fix it should continue to safely run also with winavr-gcc:

(change **[...]/tmk_core/common/keymap.c:246** to look like this one instead)
`    return (action_t){ .code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]) };`

 What do you think about it? if you think it's a fair fix you can take it over in the core or I can submit you a pull request or so.

Kr,
 tent:wq

tentator, 2016-11-11:
and by the way the ps2_mouse.c with the trackpoint configurations that set tap to click etc now also work flawlessly.. maybe I still have to understand well why most of the times when I enter bootloader mode my teensy flash tool with that green autoprogram button does not flash it...

tent:wq