tmk/tmk_keyboard #395: SLEEP_LED_ENABLE and sleep/resume problems

bjd-pfq, 2016-10-11:
Not an issue.
A note on the remark via a vis BIOS(UEFI)/Resume(Sleep&Wake)/Power cycle  problems with tmk-connected keyboards:

I am using a Sun Type 5c keyboard with tmk firmware.  At least with recent versions I experienced the following problem when having SLEEP_LED_ENABLE defined: I usually have keyclick (aka key-bell) ON on this keyboard (yes I know it might be considered a seurity issue, but hey).

During and after resuming, the keyboard would long-beep (reset) and after that keyclick would be OFF.
So it would just do a reset at RESUME time, which it should not do.  Having SLEEP_LED_ENABLE /not/ defined, the keyboard would cleany resume (no long-beep, and keyclick in its exact previous state).

No biggie, since I didn't want the SLEEP led anyway. By the way, if the SLEEP_LED  is supposed to flash, well so it did, but at a very high frequency, maybe 10x per second (guessing), i.e. it looked more like a permanent ON than a flashing LED.  Just saying.

tmk, 2016-10-12:
I dont' know it works with LED on keyboard via converter and keyboard protocol is too slow to support, in perticular for SUN 1200baud. You need LED connected to GPIO of controller directely, this function was developed originally for keyboard controller, not converter. Instead of LED on SUN keyboard you will be able to use LED on converter if you don't mind tweak some codes.

bjd-pfq, 2016-10-12:
I don't need it. Too many LEDs in the world (energy problem).
The thing of importance was that SLEEP_LED_ON caused the keyboard to reset at resume-time, which
was annoying.

tmk, 2016-10-12:
Yes, you don't need and I don't use it daily. You can just keep disable it.
I implemented this just for fun to emulate Macbook sleep breating.

BTW, It works like this if it does. :D
https://vimeo.com/61874047