tmk/tmk_keyboard #403: Infinity60 with ChibiOS drops keys

tianminc, 2016-10-29:
I flashed infinity_chibios with matrix borrowed from QMK into my Infinity60 v1.1. One thing I notice is it responses slower than the mbed version and drops keys when I type faster. I searched around and found the same issue was mentioned in #226 and fixed in #268. I have checked that in /tmk_core/protocol/chibios/main.c b/tmk_core/protocol/chibios/main.c there is no chThdSleepMilliseconds(5) after keyboard_task(), but I am still experiencing the slow scan. Any suggestions? Thank you in advance!

tmk, 2016-10-29:
You will be able to find 'bottle neck' by measuring time code spends using debug print and timer, I guess it is somewhere in file under either protocol/chibios and common/chibios.

```
uint16_t start = timer_read();
.... suspected code ...
xprintf("%dms", start - timer_read());
...

tianminc, 2016-10-30:
Thank you! I use this technique and found the bottle neck seems to be `matrix_scan()` itself. Each call to `matrix_scan()` in `keyboard.c` takes 9ms. That does not sound very right. I did not modify anything other than `matrix.c` in infinity_chibios. My matrix is get from [QMK](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/infinity60/matrix.c).

I don't have the previous version Infinity PCB but many people seem to have no problem with it. Why would changing matrix along cause such performance difference? Also just to make sure my PCB isn't broken or something, I tried the mbed version and it does not have this problem.

tmk, 2016-10-30:
I don't have problem with infinity_chibios/ and my 'prototype' infinity60, no drop key even with 'rubbing keys' test. Ah, you are using v1.1 and TMK doesn't support it yet.

Why don't  you ask QMK community or author changed matrix_scan() code?

tianminc, 2016-11-02:
Update: I tried it on my Windows 10 machine and it works without a problem even in rubbing key test. The machine that I had problem previous was running Ubuntu 14 with 3.13.0-100 kernel. That's interesting.