tmk/tmk_keyboard #404: How to use macros with UNIMAP/ACTIONMAP, please?

DreymaR, 2016-11-08:
So, I've _almost_ succeeded in making Unimap work for me, but...

I tried this, similar to what worked for me with USB-USB KEYMAP_ALL:
**#  define AC_FN9 ACTION_MACRO(TYPESTR1)              // FN9 is a user macro: Type a string**

Nope. The compiler gives an error: **TYPESTR1 undeclared here (not in a function)**

Is it because the macros used to be defined before the FN keys and now the FN keys are defined first? Or, how do I implement a macro properly with UNIMAP please?

tmk, 2016-11-08:
You have to provide macro IDs before macro definition. See this file for example.

https://github.com/tmk/tmk_keyboard/blob/093bfd6901d6153f4c3bc03d38c1bae8717065e8/keyboard/hhkb/unimap_hasu.c

DreymaR, 2016-11-09:
OK, thanks! It works now.

It'd be nice if the unimap files could be in some other directory, common for all architectures. That's part of the idea with unimap, right? :-)

closed 2016-11-15 by DreymaR