tmk/tmk_keyboard #405: logical error in xt_usb/matrix.c

bjd-pfq, 2016-11-09:
*** xt_usb/matrix.c.orig	2016-11-09 18:28:36.263834630 +0100
--- xt_usb/matrix.c	2016-11-09 18:28:57.315882187 +0100
***************
*** 162,186 ****
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;
--- 162,186 ----
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7_E0;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;

bjd-pfq, 2016-11-09:
Sorry, hope this looks more like code:

`*** xt_usb/matrix.c.orig	2016-11-09 18:28:36.263834630 +0100
--- xt_usb/matrix.c	2016-11-09 18:28:57.315882187 +0100
***************
*** 162,186 ****
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;
--- 162,186 ----
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7_E0;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;
`

bjd-pfq, 2016-11-09:
`*** xt_usb/matrix.c.orig	2016-11-09 18:28:36.263834630 +0100
--- xt_usb/matrix.c	2016-11-09 18:28:57.315882187 +0100
***************
*** 162,186 ****
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;
--- 162,186 ----
                  state = E0_2A_E0;
              else
                  state = INIT;
              break;
          case E0_2A_E0:
              if(code == 0x37)
                  matrix_make(PRINT_SCREEN);
              else
                  state = INIT;
              break;
          case E0_B7:
              if(code == 0xE0)
!                 state = E0_B7_E0;
              else
                  state = INIT;
              break;
          case E0_B7_E0:
            if(code == 0xAA)
                matrix_break(PRINT_SCREEN);
            else
                state = INIT;
            break;
          case E1:
              if (code == 0x1D)
                  state = E1_1D;
`

acid2000, 2019-11-11:
Could you put some more information here

tmk, 2019-11-12:
This is not applicable anymore and the current code base has no issue related to this.

closed 2019-11-12 by tmk