tmk/tmk_keyboard #408: Actuation Point Control

JohnCaza, 2016-11-24:
Hey man, I was wondering if maybe at some point you could take a look at how we could implement ACP (like what comes on new Topre boards) - since Topre switches are capacitive, you should be able to get a reading of "how pressed" a switch is. Would be nice to have this on the HHKB controller.

tmk, 2016-11-24:
What is actually ACP? a link plz.

Yeah, you can control actuation point with this capsense library but you have to mod or design new pcb for HHKB. It must be fun project!
https://github.com/tmk/capsense_qmatrix
https://geekhack.org/index.php?topic=76040.0

JohnCaza, 2016-11-24:
http://www.tomshardware.com/news/topre-realforce-rgb-variable-actuation,31961.html 

I wonder why we would need to redesign pcb, unless the actual controller only does 0/1, but even then at a lower level than the OS running on it, the CPU registers actuation force at a specific resistance most likely as current passes through the switch.

tmk, 2016-11-24:
My HHKB controller reads from digital signal from Topre capacitive senseor chip, which doesn't provoide  analog value at all. We have to sense capacitive switches directly ourselves to get the values.

I think Topre made new sensor chip for the new RGB Realforce.

closed 2021-07-23 by tmk