tmk/tmk_keyboard #411: obqObjectInit changed number of arguments

cubiq, 2016-12-01:
latest version of chibios changed the number of arguments for obqObjectInit

```
void obqObjectInit(output_buffers_queue_t *obqp, bool suspended, uint8_t *bp,
                   size_t size, size_t n, bqnotify_t onfy, void *link)
```

note the `bool suspended`

tmk, 2016-12-02:
I didn't update ChibiOS for a while and didn't find this change. We have to update ChibiOS-Contrib and tmk_core. I'll do this for tmk_core(tmk_keyboard) later.

Patch of tmk_core
```
diff --git a/tmk_core/protocol/chibios/usb_main.c b/tmk_core/protocol/chibios/usb_main.c
index fbf0ceb..e8f103a 100644
--- a/tmk_core/protocol/chibios/usb_main.c
+++ b/tmk_core/protocol/chibios/usb_main.c
@@ -1033,7 +1033,7 @@ void init_usb_driver(USBDriver *usbp) {
 
   chVTObjectInit(&keyboard_idle_timer);
 #ifdef CONSOLE_ENABLE
-  obqObjectInit(&console_buf_queue, console_queue_buffer, CONSOLE_EPSIZE, CONSOLE_QUEUE_CAPACITY, console_queue_onotify, (void*)usbp);
+  obqObjectInit(&console_buf_queue, true, console_queue_buffer, CONSOLE_EPSIZE, CONSOLE_QUEUE_CAPACITY, console_queue_onotify, (void*)usbp);
   chVTObjectInit(&console_flush_timer);
 #endif
 }
```

Patch of ChibiOS-Contrib
```
diff --git a/os/hal/src/hal_usb_hid.c b/os/hal/src/hal_usb_hid.c
index 56be9b7..559d290 100644
--- a/os/hal/src/hal_usb_hid.c
+++ b/os/hal/src/hal_usb_hid.c
@@ -224,7 +224,7 @@ void hidObjectInit(USBHIDDriver *uhdp) {
   ibqObjectInit(&uhdp->ibqueue, uhdp->ib,
                 USB_HID_BUFFERS_SIZE, USB_HID_BUFFERS_NUMBER,
                 ibnotify, uhdp);
-  obqObjectInit(&uhdp->obqueue, uhdp->ob,
+  obqObjectInit(&uhdp->obqueue, true, uhdp->ob,

                 USB_HID_BUFFERS_SIZE, USB_HID_BUFFERS_NUMBER,
                 obnotify, uhdp);
 }
```

tmk, 2016-12-20:
bbbb