tmk/tmk_keyboard #412: AEKII missing left and right arrow keys presses (ADB->USB)

128keaton, 2016-12-02:
I'm having a very strange issue that manifests itself in both Windows and OS X. Every couple of left or right arrow key presses, the OS doesn't respond to the presses. I checked with the hid_listen and it proved that its something with the adaptor. I don't have anything to test the keyboard with sadly, but I am using the stock ADB cable.

tmk, 2016-12-02:
>  I checked with the hid_listen and it proved that its something with the adaptor.

You mean, hid_listen shows your converter recognizes the *missing* key press from keyboard but it fails to send it to PC? Or your converter doesn't recognize the key press at all?

128keaton, 2016-12-02:
@tmk the converter doesn't recognize the keys

tmk, 2016-12-02:
OK. Then I'd recommend to replace switch on the keys, first.

closed 2016-12-02 by 128keaton