tmk/tmk_keyboard #415: Magic Key on NeXT non-adb converter

scramble45, 2016-12-10:
I'm trying to get magic key or boot magic working but not having much luck after modifying my config.h and enabling it. I hold down LShift+RShift and try H or N because I'm really trying to get NKRO working but nothing happens. Boot magic doesnt seem to work either. Is this a known issue of using a converter or am I just doing something stupidly wrong?

tmk, 2016-12-10:
Boot magic won't work with protocol converters including NeXT because the covnerters cannot recognize key events before initializing keyboard. I'd reccomend to disable boot magic build option for converters.

And unfortunately NeXT is 2KRO and rather lame in terms of key rollover, it can always send only two keys except for modifiers. NKRO doesn't make sense with NeXT at all.

scramble45, 2016-12-10:
Perfect, that is all I needed to know that clears up my suspicions. Thanks Hasu

closed 2016-12-10 by scramble45