tmk/tmk_keyboard #420: Compiling 4704 firmware without CONSOLE_ENABLE set to 'yes' causes

sorniva, 2016-12-28:


tmk, 2016-12-28:
What's your OS? If it is Windows can you try 2, 3 or 4 of this FAQ entry? That build option cuases change in USB endpoint configuration, which suffers from unexpected weirdness in Windows.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Build#edit-configuration-but-not-change

sorniva, 2016-12-28:
OS is Debian, have tried on two different Debian machines, with and without USB hub.

tmk, 2016-12-28:
OK. I'll look into later.

tmk, 2017-06-22:
I confirmed TMK IBM4704 converter with my 62key model spits single 'd' when configured with `CONSOLE_ENABLE = no`, though the 'd' is not reapeated for some reason.

This problem is caused by 'keyboard_id' byte that 4704 keyboards send at startup, when console is enabled it is read and displayed, while it is not consumed at startup when disabled and the byte is handled as keycode wrongly later.

`xprintf()` is not removed and the `keyboard_id` is not read when console is disabled.
https://github.com/tmk/tmk_keyboard/blob/master/converter/ibm4704_usb/matrix.c#L83

Id of 62key is A3 and Alps model is A6, those ids are identical to keycode of 'd' and 'h' respectively.
https://github.com/tmk/tmk_keyboard/blob/master/converter/ibm4704_usb/ibm4704.txt#L27-L37

This fixes the problem.
```
diff --git a/converter/ibm4704_usb/matrix.c b/converter/ibm4704_usb/matrix.c
index c02d9e5..acba17d 100644
--- a/converter/ibm4704_usb/matrix.c
+++ b/converter/ibm4704_usb/matrix.c
@@ -80,7 +80,8 @@ void matrix_init(void)
     print("IBM 4704 converter\n");
     matrix_clear();
     _delay_ms(2000);    // wait for keyboard starting up
-    xprintf("Keyboard ID: %02X\n", ibm4704_recv());
+    uint8_t keyboard_id = ibm4704_recv();
+    xprintf("Keyboard ID: %02X\n", keyboard_id);
     enable_break();
 }
```

closed 2017-06-22 by tmk