tmk/tmk_keyboard #427: No rule to make target `protocol/lufa/LUFA-git/LUFA/Build/lufa_sources.mk' - Error

dunkarooftop, 2017-01-17:
OSX El Capitan
Fresh download didn't even modify any file yet get this error when try to "make -f Makefile"

```
[Dunk:...yboard-master/keyboard/hhkb]$ make -f makefile
../../tmk_core/protocol/lufa/LUFA-git/LUFA/makefile:33: protocol/lufa/LUFA-git/LUFA/Build/lufa_sources.mk: No such file or directory
make: *** No rule to make target `protocol/lufa/LUFA-git/LUFA/Build/lufa_sources.mk'.  Stop.
```

Trying to flash my HHKB. I was using the older version (2015/04/22) and have no problem at all but all the sudden I get this error message

```
In file included from ../../tmk_core/common/action.h:24:0,
                 from keymap_common.h:23,
                 from keymap_dunk.c:6:
keymap_dunk.c: In function 'keymap_fn_to_action':
../../tmk_core/common/action_code.h:184:41: error: expected expression before '{' token
 #define ACTION_NO                       { .code = 0 }
                                         ^
keymap_dunk.c:356:19: note: in expansion of macro 'ACTION_NO'
     action.code = ACTION_NO;
                   ^
../../tmk_core/common/action_code.h:184:41: error: expected expression before '{' token
 #define ACTION_NO                       { .code = 0 }
                                         ^
keymap_dunk.c:376:24: note: in expansion of macro 'ACTION_NO'
     if (action.code == ACTION_NO && FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
                        ^
make: *** [obj_hhkb_lufa/keymap_dunk.o] Error 1
```

 That's why I wanna try see if the new build will fix it. Thank you

tmk, 2017-01-17:
Download source and these command should work unless you change anything.
```
$ cd tmk_keyboard-master/keyboard/hhkb
$ make -f Makefile
```

I think `make -f makefile` should output following error on decent system but I don't know much about OSX. Anyway, you should use `make -f Makefile`.
```
$ make -f makefile
make: makefile: No such file or directory
make: *** No rule to make target 'makefile'.  Stop.
````

tmk, 2017-01-17:
And updating your toolchain may help.

dunkarooftop, 2017-01-17:
I just clone the source again and use ```make -f Makefile``` and it's still give me the same error.

The toolchain is CrossPack for AVR latest version (which have not been updated since 2013)

Thanks for the fast reply hasu :) any more ideas for troubleshoot?

dunkarooftop, 2017-01-17:
I notice it's complaining ``` protocol/lufa/LUFA-git/LUFA/Build/lufa_sources.mk' ``` from termainal

and i went into Finder and see my local file name for that file is ```lufa sources.mk``` instead of ``` lufa_sources.mk``` with ```space``` instead of ```_``` dose it matter ?

tmk, 2017-01-17:
It matters. Weird, I don't know why this file name problem happens.

dunkarooftop, 2017-01-18:
I found out what's wrong now, I have Hazel (an OSX utility) to watch my download folder to clean up file name. And I extract in that folder so Hazel rename some files. My bad

Thank you again for your amazing work, my fingers love you. Get your controller for my HHKB.

dunkarooftop, 2017-01-18:
After I try to compile my old mod still get these error. It used to work before, I wonder if the toolchain get outdated.

```
[Dunk:...yboard-master/keyboard/hhkb]$ make -f Makefile KEYMAP=dunk

-------- begin --------
avr-gcc (GCC) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mkdir -p obj_hhkb
Compiling C: keymap_dunk.c
avr-gcc -c -mmcu=atmega32u4 -gdwarf-2 -DF_CPU=16000000UL -DBOOTLOADER_SIZE=4096 -DF_USB=16000000UL -DARCH=ARCH_AVR8 -DUSB_DEVICE_ONLY -DUSE_FLASH_DESCRIPTORS -DUSE_STATIC_OPTIONS="(USB_DEVICE_OPT_FULLSPEED | USB_OPT_REG_ENABLED | USB_OPT_AUTO_PLL)" -DFIXED_CONTROL_ENDPOINT_SIZE=8  -DFIXED_NUM_CONFIGURATIONS=1 -DPROTOCOL_LUFA -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DNKRO_ENABLE -DVERSION=unknown -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_hhkb/keymap_dunk.lst -I. -I../../tmk_core -I../../tmk_core/protocol/lufa -I../../tmk_core/protocol/lufa/LUFA-git -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_hhkb_keymap_dunk.o.d  keymap_dunk.c -o obj_hhkb/keymap_dunk.o
In file included from /usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/include/stdint.h:9:0,
                 from keymap_common.h:20,
                 from keymap_dunk.c:6:
/usr/local/Cellar/avr-gcc/4.9.3/avr/include/stdint.h:125:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'typedef'
 typedef signed int int8_t __attribute__((__mode__(__QI__)));
 ^
/usr/local/Cellar/avr-gcc/4.9.3/avr/include/stdint.h:163:9: error: unknown type name 'int8_t'
 typedef int8_t   int_least8_t;
         ^
/usr/local/Cellar/avr-gcc/4.9.3/avr/include/stdint.h:217:9: error: unknown type name 'int8_t'
 typedef int8_t int_fast8_t;
         ^
In file included from ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/AudioClass.h:71:0,
                 from ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/USB.h:412,
                 from ../../tmk_core/protocol/lufa/descriptor.h:47,
                 from ../../tmk_core/common/report.h:83,
                 from ../../tmk_core/common/action_util.h:21,
                 from keymap_common.h:27,
                 from keymap_dunk.c:6:
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:252:18: error: unknown type name 'int8_t'
    static inline int8_t Audio_Device_ReadSample8(USB_ClassInfo_Audio_Device_t* const AudioInterfaceInfo)
                  ^
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:254:18: error: unknown type name 'int8_t'
    static inline int8_t Audio_Device_ReadSample8(USB_ClassInfo_Audio_Device_t* const AudioInterfaceInfo)
                  ^
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h: In function 'Audio_Device_ReadSample8':
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:256:5: error: unknown type name 'int8_t'
     int8_t Sample;
     ^
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h: At top level:
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:327:55: error: unknown type name 'int8_t'
                                                 const int8_t Sample) ATTR_NON_NULL_PTR_ARG(1) ATTR_ALWAYS_INLINE;
                                                       ^
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:329:55: error: unknown type name 'int8_t'
                                                 const int8_t Sample)
                                                       ^
In file included from ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/HIDClassDevice.h:58:0,
                 from ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/HIDClass.h:72,
                 from ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/USB.h:414,
                 from ../../tmk_core/protocol/lufa/descriptor.h:47,
                 from ../../tmk_core/common/report.h:83,
                 from ../../tmk_core/common/action_util.h:21,
                 from keymap_common.h:27,
                 from keymap_dunk.c:6:
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/../Common/HIDClassCommon.h:654:4: error: unknown type name 'int8_t'
    int8_t  X; /**< Current delta X movement of the mouse. */
    ^
../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Device/../Common/HIDClassCommon.h:655:4: error: unknown type name 'int8_t'
    int8_t  Y; /**< Current delta Y movement on the mouse. */
    ^
In file included from ../../tmk_core/common/action_util.h:21:0,
                 from keymap_common.h:27,
                 from keymap_dunk.c:6:
../../tmk_core/common/report.h:147:5: error: unknown type name 'int8_t'
     int8_t x;
     ^
[... 37 lines cut ...]
.
.
.
   ^
keymap_dunk.c:148:3: warning: (near initialization for 'fn_actions_4[0]')
keymap_dunk.c:148:3: error: field name not in record or union initializer
keymap_dunk.c:148:3: error: (near initialization for 'fn_actions_4[0]')
In file included from ../../tmk_core/common/action.h:24:0,
                 from keymap_common.h:23,
[... cut ...]
keymap_dunk.c:356:19: note: in expansion of macro 'ACTION_NO'
     action.code = ACTION_NO;
                   ^
../../tmk_core/common/action_code.h:187:41: error: expected expression before '{' token
 #define ACTION_NO                       { .code = 0 }
                                         ^
keymap_dunk.c:376:24: note: in expansion of macro 'ACTION_NO'
     if (action.code == ACTION_NO && FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
                        ^
make: *** [obj_hhkb/keymap_dunk.o] Error 1
```

dunkarooftop, 2017-01-18:
avrdude --help GNU Make 3.81
avr-gcc (GCC) 4.9.3
avrdude version 6.0.1,

closed 2017-01-18 by dunkarooftop

tmk, 2017-01-18:
It's Not too old tools at all but do you have proper avr libc? Seems like People on mac use home brew to install toolchain recently. Try  brew if current toolchain doesn't work well.

In this commit e5cb846 action code changed just integer to struct, this probably caused part of the errors.
And see this commit c04b6be which changed type of fn_action, you have to use action_t instead of uint16_t.

dunkarooftop, 2017-01-18:
fix most of the error now I only have 1 error left about the "ACTION_NO"

```
/*
 * translates Fn keycode to action
 * for some layers, use different translation table
 */
action_t keymap_fn_to_action(uint8_t keycode)
{
    uint8_t layer = biton32(layer_state);

    action_t action;
    action.code = ACTION_NO;

    if (layer == 1 && FN_INDEX(keycode) < FN_ACTIONS_1_SIZE) {
        action.code = pgm_read_word(&fn_actions_1[FN_INDEX(keycode)]);
    }

    if (layer == 2 && FN_INDEX(keycode) < FN_ACTIONS_2_SIZE) {
        action.code = pgm_read_word(&fn_actions_2[FN_INDEX(keycode)]);
    }

    if (layer == 3 && FN_INDEX(keycode) < FN_ACTIONS_3_SIZE) {
        action.code = pgm_read_word(&fn_actions_3[FN_INDEX(keycode)]);
    }

    if (layer == 4 && FN_INDEX(keycode) < FN_ACTIONS_4_SIZE) {
        action.code = pgm_read_word(&fn_actions_4[FN_INDEX(keycode)]);
    }
    // by default, use fn_actions from default layer 0
    // this is needed to get mapping for same key, that was used switch to some layer,
    // to have possibility to switch layers back
    if (action.code == ACTION_NO && FN_INDEX(keycode) < FN_ACTIONS_SIZE) {
        action.code = pgm_read_word(&fn_actions[FN_INDEX(keycode)]);
    }

    return action;
}
```

What should I replace the 2 "ACTION_NO" from the code? Sorry I don't know C most of my keymapping is hacking from other people's code.

dunkarooftop, 2017-01-18:
never mind I tried "KC_NO" and it worked !!! Thank you so much hasu !!!