tmk/tmk_keyboard #428: 'Double tap repeat' on tap key

tmk, 2017-01-17:
A use case that 'Double tap repeat' prevents user intention.
https://geekhack.org/index.php?topic=41989.msg2342065#msg2342065

kdewilde, 2017-01-18:
I did some tests.

If i tap spacebar and then hold it outputs 2 spaces and then there are 2 scenarios

1. if i start pressing buttons it outputs those (for example ijkl)
2. if i just keep holding spacebar it starts to repeat spacebar

I don't know much about C but i'm willing to check this out. How can i put a break there and how do i build C?
I'm running it on a virtualbox linux and i don't know much about linux :) so if you know how to do what i asked on linux that would be a plus.

samuelngsh, 2017-02-04:
I use the semicolon key as a ACTION_LAYER_TAP_KEY to activate another layer when held down (hjkl for arrow key, etc), while a single tap on it produces a normal semicolon. 

I'm having the same issue as mentioned in the linked thread as well as by kdewilde's comment. If I type ";", then very quickly hold down ";" again with the intention to activate the layer, what happens instead is I'll get extra ";"s being printed on the window. 

In my own use case, I usually don't have to type ; or : in quick succession, therefore I made the following quick modification in action.c starting from line 290 to temporarily fix double tap problem for my own use case:

`                    /* tap key */
                    if (event.pressed) {
                        if (tap_count == 1) {
                            dprint("KEYMAP_TAP_KEY: Tap: register_code\n");
                            register_code(action.layer_tap.code);
                        } else if (tap_count > 1) {
                            dprint("KEYMAP_TAP_KEY: No tap: On on press\n");
                            layer_on(action.layer_tap.val);
                        } else {
                            dprint("KEYMAP_TAP_KEY: No tap: On on press\n");
                            layer_on(action.layer_tap.val);
                        }   
                    } else {
                        if (tap_count == 1) {
                            dprint("KEYMAP_TAP_KEY: Tap: unregister_code\n");
                            unregister_code(action.layer_tap.code);
                        } else if (tap_count > 1) {
                            dprint("KEYMAP_TAP_KEY: No tap: Off on release\n");
                            layer_off(action.layer_tap.val);
                        } else {
                            dprint("KEYMAP_TAP_KEY: No tap: Off on release\n");
                            layer_off(action.layer_tap.val);
                        }   
                    }`

This makes it such that the first tap of the modifier produces the predetermined value, while subsequent activations of the modifier always activates the function layer. In my case, the first tap of ";" produces the semicolon, while subsequent holds always activate the function layer.

Note that if you need to tap the modifier multiple times (in my case, if I need to type ";;;;;;;;"), you will have to wait until the TAPPING_TERM (config.h in your keymap folder) times out before you tap the next one, which means ; *wait* ; *wait* ; *wait*...

This is a non-ideal temporary solution that fits my own use for now, I'm in the middle of many other deadlines so I can't dig any deeper into this for a more elegant fix. Let's see what hasu or other contributors can come up with!