tmk/tmk_keyboard #438: Two keys doing the same thing on e RS68 II+ rev C

dotchoucou, 2017-02-12:
Hi,

I have a problem with my RS68 II+ rev C.

Both red keys on this layout does the same thing when I press them:
http://www.keyboard-layout-editor.com/#/gists/462f442185cbc90095344fe50ecc9261
i.e. the right one behaves like the left one although the layout as shown here is different.

Can anyone help?
Is that a layout configuration problem?

kairyu, 2017-02-12:
I think you are asking in wrong place.
Here is a similar discussion: https://geekhack.org/index.php?topic=69282.msg1673980#msg1673980

dotchoucou, 2017-02-12:
Maybe its not about the keyboard, but the way I program it, that's why I came here.

dotchoucou, 2017-02-12:
I answer to myself thanks to someone on geekhack :
needed to use "iso |" in the layout.

Working now; :)

closed 2017-02-12 by tmk