tmk/tmk_keyboard #441: Prevent Stuck Modifiers

dionmunk, 2017-02-23:
I am experiencing an issue where a modifier key gets stuck when switching from a momentary layer back to the base layer. The key in question is mapped to A on Layer0 and LCTRL on Layer1.

Would it be possible to get a PREVENT_STUCK_MODIFIERS option like QMK?

It is referenced:
https://github.com/qmk/qmk_firmware/issues/579
https://github.com/qmk/qmk_firmware/pull/182

DreymaR, 2017-02-27:
I'd like to make a better layer switch function instead (I have an issue on that here). A temporary layer switch that restores modifier states on release would be ideal for my purposes, and wouldn't lead to an overhead per key like QMK's solution does. Or just an optional option for the existing layer switches.

dionmunk, 2017-02-27:
If I could figure out how to have it release CTRL when I release FN0, that would actually solve my problem. I'm not adept enough to program a function to do this though.

tmk, 2017-02-28:
Can anyone port this PR to TMK and test it?
https://github.com/qmk/qmk_firmware/pull/182

DreymaR, 2017-03-01:
Is that a good fix though? I'm not sure I understand it entirely, but it seems that they store every pressed key which uses up a lot of memory. Surely there must be better ways of handling modifier jam? (Some are suggested in the comments, I think?)

tmk, 2017-03-08:
@DreymaR I think the PR seems to me be ideal and straight solution for the key stuck problem and what I've wanted for long time. Yes,  you are right, it costs more memory and time in return. But if the cost is acceptable I want to merge it. I need universal solution and don't want ad hoc solution moreover for the problem.

DreymaR, 2017-03-08:
I still don't see the rationale for this solution. In my head:
• You enter a layer, and may change modifier states in that layer
• When you leave the layer, you may not want these changed states (are there exceptions to this?)
• Therefore, the best solution is to make the modifiers on that layer "weak" so they are lost on leaving

tmk, 2017-05-30:
Fixed at 	ba2883f

closed 2017-05-30 by tmk