tmk/tmk_keyboard #442: Question: Inverting shift behaviour for one key

chrisbarrett, 2017-02-26:
I would like to invert the behaviour of the semicolon key, such that it outputs `:` with no modifiers and `;` when shift is held. I've followed this [link][1] on the FAQ, but I don't see how the discussion or C example code relates to this. 

Can you tell me how to achieve this? Any examples in C would be cool too. 😸 Thanks for the great work!

[1]: https://github.com/qmk/qmk_firmware/wiki/Keymap-examples#reverse-shifted-for-numbers

DreymaR, 2017-02-27:
Okay, so that's technically QMK and it may have some changes from TMK in this respect.

This task is doable without C coding I think, but it'd get clunky. I guess it'd be better to make a user-function to switch shift behaviour for one key.

I have quite a few macro keys that produce Shift+key or Ctrl+key, But to get the unshifted key when Shift is held down, you'd have to define a shifted layer manually, and that'd get ugly afaik.

chrisbarrett, 2017-02-27:
I feel like a doofus for looking at the faq for the wrong project. 😅 Thanks for the response. Sounds like studying the C customisation options are my best bet. 👍

closed 2017-02-27 by chrisbarrett