tmk/tmk_keyboard #445: USB-USB converter not working with Pro Micro 3.3V 8MHz

savf, 2017-02-28:
I tried building a USB-USB converter as outlined [in this post](https://geekhack.org/index.php?topic=80421.0), but the Pro Micro with the usb_usb hex file flashed to it does not get recognized consistently by any computer (tested on Ubuntu Linux 16.04 and macOS Sierra).

Sometimes it works and the converter gets recognized correctly, but most of the time I get the following output in syslog on Ubuntu Linux 16.04:

```
Feb 28 19:32:01 mianyang kernel: [  295.364131] usb 3-1: new full-speed USB device number 6 using uhci_hcd
Feb 28 19:32:01 mianyang kernel: [  295.488096] usb 3-1: device descriptor read/64, error -71
Feb 28 19:32:01 mianyang kernel: [  295.712087] usb 3-1: device descriptor read/64, error -71
Feb 28 19:32:01 mianyang kernel: [  295.932059] usb 3-1: new full-speed USB device number 7 using uhci_hcd
Feb 28 19:32:02 mianyang kernel: [  296.052132] usb 3-1: device descriptor read/64, error -71
Feb 28 19:32:02 mianyang kernel: [  296.276117] usb 3-1: device descriptor read/64, error -71
Feb 28 19:32:02 mianyang kernel: [  296.492137] usb 3-1: new full-speed USB device number 8 using uhci_hcd
Feb 28 19:32:02 mianyang kernel: [  296.908054] usb 3-1: device not accepting address 8, error -71
Feb 28 19:32:03 mianyang kernel: [  297.020127] usb 3-1: new full-speed USB device number 9 using uhci_hcd
Feb 28 19:32:03 mianyang kernel: [  297.436079] usb 3-1: device not accepting address 9, error -71
Feb 28 19:32:03 mianyang kernel: [  297.436113] usb usb3-port1: unable to enumerate USB device
```

The output from the console in macOS Sierra is:

```
default	19:42:25.406137 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff803fb7fe70 failed (0xe00002e2)
default	19:42:27.551695 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff803fb800a0 failed (0xe00002e2)
default	19:42:29.696761 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff8036b67f10 failed (0xe00002e2)
default	19:42:31.842265 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff80539c62a0 failed (0xe00002e2)
default	19:42:33.987416 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff803ed97340 failed (0xe00002e2)
default	19:42:37.156529 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff8053a58c40 failed (0xe00002e2)
default	19:42:42.374262 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff803fb801f0 failed (0xe00002e2)
default	19:42:51.688441 +0100	kernel	AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe 0xffffff803fb801f0 failed (0xe00002e2)
default	19:42:51.688503 +0100	kernel	032451.311144 AppleUSBUHCIPort@1d100000: AppleUSBHostPort::disconnect: persistent enumeration failures
```

tmk, 2017-02-28:
It smels like hardware issue, resolder all joints again.
Also I'm not sure whether it works at 8MHz, better use 16MHz.

savf, 2017-02-28:
First off: Thank you very much for the fast reply! I tested it without having it soldered to the mini USB host shield, just the Pro Micro as I received it. I tried it on a second identical Pro Micro (also 3.3V and 8MHz) and it did also not work there (same errors).

You suggested using 16MHz: Do you mean running the Pro Micro with 3.3V but at 16MHz or do you suggest using a Pro Micro with 5V and 16MHz? Wouldn't that be a problem with the mini host shield which needs 3.3V to be powered?

tmk, 2017-02-28:
First of all, sodler them properly. Without sodlering  I don't think you can get good connections.

I would recommend Leonardo and Circuit@home genuine shield as first post of my thread says.
https://geekhack.org/index.php?topic=69169.0

savf, 2017-02-28:
I wanted to only try the Pro Micro without having anything soldered or connected, just to see whether the computer would recognize the Pro Micro with the firmware flashed to it, when I plug it in.
After soldering everything together according to the thread I initially posted, it magically seems to work now, I don't understand why the computer wouldn't recognize the Pro Micro without the host shield, but as soon as I soldered the host shield, it recognizes it every time I plug in (so far). Well I hope this keeps working and **thank you again for the quick help**!

closed 2017-02-28 by savf