tmk/tmk_keyboard #448: Feature: ACTION_LAYER_TAP_KEY with mods

eulerdisk, 2017-03-06:
I would really like to have an `ACTION_LAYER_TAP_KEY` with mods.
Something like this:
`ACTION_LAYER_TAP_KEY(2, KC_LEFT, MY_MODS)`
If you tap the key you have `KC_LEFT`. If you hold the key you have momentary switch to layer 2 with `MY_MODS` activated.
Is that something doable with the current firmware implementation ?

I could implement that myself, maybe with a bit of mentoring.

tmk, 2017-03-07:
It is difficult to implement built-in action like that because you have to pack the parameters into 16 bit action code. Layer parameter takes 5bits and keycode 8bits and mods 8bits(or 4bits), 16bits doesn't seem to be enough for those.

But you can still implement you own action function in keymap file.

tmk, 2017-03-07:
And see this entry too.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#how-to-define-your-own-actionhow-to-explore-build-in-action

DreymaR, 2017-03-08:
Actually, wouldn't it be possible to allow everything from ACTION_KEY up to take this kind of syntax:
`ACTION_KEY(KC_#### | MOD_####)`
`ACTION_LAYER_TAP_KEY(#, KC_#### | MOD_####)`
etc

This would alleviate the need for several functions. Would it be possible?

tmk, 2017-03-08:
I'm not sure I understand what you meant fully but I dont' think so.
4 bits of action code is used to identify action kind, 
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_code.h#L102-L122

and 12 bits is left for parameters. This comment describes how existent actions use the bits for reference.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_code.h#L20-L101

The 12 bits are not enough to define `ACTION_LAYER_TAP_KEY(#, KC_#### | MOD_####)` as I wrote before.

We will need new codes and framework to meet this request.

DreymaR, 2017-03-08:
Oh, sorry. I did not realize about the 4 bits of action code. Scratch that, then! And thanks for the explanation! 👍 

However: What is the reason to have both `ACTION_KEY` and `ACTION_MODS_KEY`? In this case only 12 bits would be needed and the above suggestion should cover both functions, unless I'm mistaken?

tmk, 2017-03-08:
`ACTION_MODS_KEY(0, KC_A)` and  `ACTION_KEY(KC_A)` are identical in terms of code bits, just for easiness to define.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_code.h#L217-L219

Ah, I see. You want to define `ACTION_KEY(KC_A | MOD_LSFT)` instead of `ACTION_MODS_KEY(MOD_LSFT, KC_A)` ? It makes sense. This is possible with some changes in `action_codes.h`.

I didn't test yet but this patch will work.
```
diff --git a/tmk_core/common/action_code.h b/tmk_core/common/action_code.h
index 4b66428..1eb6910 100644
--- a/tmk_core/common/action_code.h
+++ b/tmk_core/common/action_code.h
@@ -200,26 +200,26 @@ typedef union {
  *   bit 4      +----- LR flag(Left:0, Right:1)
  */
 enum mods_bit {
-    MOD_NONE = 0x00,
-    MOD_LCTL = 0x01,
-    MOD_LSFT = 0x02,
-    MOD_LALT = 0x04,
-    MOD_LGUI = 0x08,
-    MOD_RCTL = 0x11,
-    MOD_RSFT = 0x12,
-    MOD_RALT = 0x14,
-    MOD_RGUI = 0x18,
+    MOD_NONE = 0x0000,
+    MOD_LCTL = 0x0100,
+    MOD_LSFT = 0x0200,
+    MOD_LALT = 0x0400,
+    MOD_LGUI = 0x0800,
+    MOD_RCTL = 0x1100,
+    MOD_RSFT = 0x1200,
+    MOD_RALT = 0x1400,
+    MOD_RGUI = 0x1800,
 };
 enum mods_codes {
     MODS_ONESHOT = 0x00,
     MODS_TAP_TOGGLE = 0x01,
 };
 #define ACTION_KEY(key)                 ACTION(ACT_MODS, (key))
-#define ACTION_MODS(mods)               ACTION(ACT_MODS, ((mods)&0x1f)<<8 | 0)
-#define ACTION_MODS_KEY(mods, key)      ACTION(ACT_MODS, ((mods)&0x1f)<<8 | (key))
-#define ACTION_MODS_TAP_KEY(mods, key)  ACTION(ACT_MODS_TAP, ((mods)&0x1f)<<8 | (key))
-#define ACTION_MODS_ONESHOT(mods)       ACTION(ACT_MODS_TAP, ((mods)&0x1f)<<8 | MODS_ONESHOT)
-#define ACTION_MODS_TAP_TOGGLE(mods)    ACTION(ACT_MODS_TAP, ((mods)&0x1f)<<8 | MODS_TAP_TOGGLE)
+#define ACTION_MODS(mods)               ACTION(ACT_MODS, ((mods)&0x1f00) | 0)
+#define ACTION_MODS_KEY(mods, key)      ACTION(ACT_MODS, ((mods)&0x1f00) | (key))
+#define ACTION_MODS_TAP_KEY(mods, key)  ACTION(ACT_MODS_TAP, ((mods)&0x1f00) | (key))
+#define ACTION_MODS_ONESHOT(mods)       ACTION(ACT_MODS_TAP, ((mods)&0x1f00) | MODS_ONESHOT)
+#define ACTION_MODS_TAP_TOGGLE(mods)    ACTION(ACT_MODS_TAP, ((mods)&0x1f00) | MODS_TAP_TOGGLE)
 
 
 /*
@@ -271,7 +271,7 @@ enum layer_pram_tap_op {
 #define ACTION_LAYER_ON_OFF(layer)                  ACTION_LAYER_TAP((layer), OP_ON_OFF)
 #define ACTION_LAYER_OFF_ON(layer)                  ACTION_LAYER_TAP((layer), OP_OFF_ON)
 #define ACTION_LAYER_SET_CLEAR(layer)               ACTION_LAYER_TAP((layer), OP_SET_CLEAR)
-#define ACTION_LAYER_MODS(layer, mods)              ACTION_LAYER_TAP((layer), 0xc0 | ((mods)&0x1f
))
+#define ACTION_LAYER_MODS(layer, mods)              ACTION_LAYER_TAP((layer), 0xc0 | ((mods>>8)&0
x1f))
 /* With Tapping */
 #define ACTION_LAYER_TAP_KEY(layer, key)            ACTION_LAYER_TAP((layer), (key))
 #define ACTION_LAYER_TAP_TOGGLE(layer)              ACTION_LAYER_TAP((layer), OP_TAP_TOGGLE)
```

DreymaR, 2017-03-08:
Indeed. This definition of mods seems more consistent to me. But it's you who see the large picture of  course! :-)

eulerdisk, 2017-03-09:
@tmk I see now the 16 bit problem digging into the code.
I will probably try the `action_function` route.
Thanks.

closed 2017-05-04 by tmk