tmk/tmk_keyboard #449: Make comments on keyboard/teensy_lc_onekey/ makefiles consistent

andrumar10, 2017-03-07:
[Makefile]( https://github.com/tmk/tmk_keyboard/blob/master/keyboard/teensy_lc_onekey/Makefile)  in teensy_lc_onekey has:
```
# Cortex version
#Teensy LC is cortex-m0plus; Teensy 3.x are cortex-m4
MCU  = cortex-m0plus
```

While [Makefile.3.0]( https://github.com/tmk/tmk_keyboard/blob/master/keyboard/teensy_lc_onekey/Makefile.3.0) and [3.2](https://github.com/tmk/tmk_keyboard/blob/master/keyboard/teensy_lc_onekey/Makefile.3.2)  in teensy_lc_onekey has:
```
# Cortex version
#Teensy LC is cortex-m0; Teensy 3.x are cortex-m4
MCU  = cortex-m0plus
```
Both should be cortex-m0plus I think.
small fix but it'd make my day better :satisfied:

tmk, 2017-03-07:
Makefile.3.0 and 3.2 target for Teensy3.0 and 3.2 respectively, I think it is correct.

tmk, 2017-03-07:
By the way, `teensy_arm_onekey` is better for name of the project because it also supports Teensy3.x not only LC.

andrumar10, 2017-03-07:
oohh ok! Sorry I'm still working my way though the code, thanks for the clarification!
and yeah `teensy_arm_onekey` makes more sense

closed 2017-03-07 by andrumar10