tmk/tmk_keyboard #456: Problem with my function for "hyper" key

dunkarooftop, 2017-04-08:
I have a function for when tap send ```:``` and when held with other key send ```hyper``` but sometimes some modifier key will not release when in hyper key mode after release the physical key. The only fix is to reboot. Can someone take a look at my hack see what I did wrong. Thanks :)

```
        case HYPER_COLON:
            if (record->event.pressed) {
                if (record->tap.count > 0 && !record->tap.interrupted) {
                    if (record->tap.interrupted) {
                        dprint("tap interrupted\n");
                        add_weak_mods(MOD_BIT(KC_LGUI));
                        add_weak_mods(MOD_BIT(KC_LALT));
                        add_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                        add_weak_mods(MOD_BIT(KC_LSHIFT));
                        send_keyboard_report();
                    }
                  } else {
                        add_weak_mods(MOD_BIT(KC_LGUI));
                        add_weak_mods(MOD_BIT(KC_LALT));
                        add_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                        add_weak_mods(MOD_BIT(KC_LSHIFT));
                        send_keyboard_report();
                }
            } else {
                if (record->tap.count > 0 && !(record->tap.interrupted)) {
                    add_weak_mods(MOD_BIT(KC_LSHIFT));
                    send_keyboard_report();
                    register_code(KC_SCOLON);
                    unregister_code(KC_SCOLON);
                    del_weak_mods(MOD_BIT(KC_LSHIFT));
                    send_keyboard_report();
                    record->tap.count = 0;  // ad hoc: cancel tap
                } else {
                    del_weak_mods(MOD_BIT(KC_LGUI));
                    del_weak_mods(MOD_BIT(KC_LALT));
                    del_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                    del_weak_mods(MOD_BIT(KC_LSHIFT));
                    send_keyboard_report();
                }
            }
            break;
```

tmk, 2017-04-08:
This is code for `ACTION_MODS_TAP_KEY(mods, key)`, you can know how it works from this.

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action.c#L152-L177

dunkarooftop, 2017-04-09:
Thanks again!! Here is a working version in case if anyone wondering in the future.

```
        case HYPER_COLON:
            if (record->event.pressed) {
                if (record->tap.count > 0) {
                    if (record->tap.interrupted) {
                        dprint("tap interrupted\n");
                        // ad hoc: set 0 to cancel tap
                        record->tap.count = 0;
                        add_weak_mods(MOD_BIT(KC_LGUI));
                        add_weak_mods(MOD_BIT(KC_LALT));
                        add_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                        add_weak_mods(MOD_BIT(KC_LSHIFT));
                        send_keyboard_report();
                    } else {
                        dprint("MODS_TAP: Tap: register_code\n");
                        add_weak_mods(MOD_BIT(KC_LSHIFT));
                        send_keyboard_report();
                        register_code(KC_SCOLON);
                        record->tap.count = 0;  // ad hoc: cancel tap
                    }
                  } else {
                        add_weak_mods(MOD_BIT(KC_LGUI));
                        add_weak_mods(MOD_BIT(KC_LALT));
                        add_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                        add_weak_mods(MOD_BIT(KC_LSHIFT));
                        send_keyboard_report();
                }
            } else {
                if (record->tap.count > 0) {
                    unregister_code(KC_SCOLON);
                    del_weak_mods(MOD_BIT(KC_LSHIFT));
                    send_keyboard_report();
                    record->tap.count = 0;  // ad hoc: cancel tap
                } else {
                    del_weak_mods(MOD_BIT(KC_LGUI));
                    del_weak_mods(MOD_BIT(KC_LALT));
                    del_weak_mods(MOD_BIT(KC_LCTL));                                                                        
                    del_weak_mods(MOD_BIT(KC_LSHIFT));
                    send_keyboard_report();
                }
            }
            break;
```

closed 2017-04-09 by dunkarooftop