tmk/tmk_keyboard #458: weak_mods and real_mods

tmk, 2017-04-11:
`ACTION_KEY(KC_LSFT)` is registered in `real_mods` while `ACTION_KEY(MOD_LSFT)` is registered in `weak_mods`.
This difference is useful? or both can used `real_mod`? 

These actions should be identical?
```
ACTION_KEY(KC_LSFT)
ACTION_KEY(MOD_LSFT)
ACTION_MODS(MOD_LSFT)
ACTION_MODS_KEY(MOD_LSFT, KC_NO)
```

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action.c#L78-L97
```
        case ACT_LMODS:
        case ACT_RMODS:
            {
                uint8_t mods = (action.kind.id == ACT_LMODS) ?  action.key.mods :
                                                                action.key.mods<<4;
                if (event.pressed) {
                    if (mods) {
                        add_weak_mods(mods);
                        send_keyboard_report();
                    }
                    register_code(action.key.code);
                } else {
                    unregister_code(action.key.code);
                    if (mods) {
                        del_weak_mods(mods);
                        send_keyboard_report();
                    }
                }
            }
            break;
```

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action.c#L517-L530
```
void clear_keyboard_but_mods(void)
{
    clear_weak_mods();
    clear_keys();
    send_keyboard_report();
#ifdef MOUSEKEY_ENABLE
    mousekey_clear();
    mousekey_send();
#endif
#ifdef EXTRAKEY_ENABLE
    host_system_send(0);
    host_consumer_send(0);
#endif
}
```

dionmunk, 2017-04-11:
I think this difference is useful. This would solve the issue I have in https://github.com/tmk/tmk_keyboard/issues/441.

tmk, 2017-04-11:
Assuming you are using `ACTION_MODS(MOD_LCTL)` on layer 1 to prevent the key stuck. But once key stuck problem is fixed by implementing `PREVENT_STUCK_MODIFIERS`  you won't need `weak_mods` and will be able to use usual ACTION_KEY(KC_LCTL).

After the fix `weak_mods` is still needed by keymap Macro and Function and you can access it through these API.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_util.h#L44-L48

dionmunk, 2017-04-11:
I was not using `ACTION_MODS(MOD_LCTL)`, but I just put that into my keymap and tested it. It does exactly what I wanted. Thanks!