tmk/tmk_keyboard #464: Can't assign key, other weird Keymap Editor issues with IBM4704

NFiorentini, 2017-04-24:
I have been unable to assign anything to the key on the 4th row, last column [image] using the Keymap Editor.

However, the key does show output with hid_listen ...
[image]

Also, when using a customized .hex generated by the Keymap Editor, the keyboard will become unresponsive after several minutes of inactivity until there have been several keypresses in rapid succession.  And I've had keys that were working one moment suddenly stop working later, but start working again if I reflash the .hex file or unplug/replug in the keyboard.

I'm using a Teensy2.0 with the Teensy gui to flash, on Windows 10.

tmk, 2017-04-26:
I don't have alps board and can't confirm it myself unfortunately. With default firmware the key spits out F16, couldn't you see F16 with tool like aquakey or switchhitter?

I've not hear those behaviour from TMK converter users so far. Let me know if you found bug on my firmware.

tmk, 2017-04-26:
It is likely to locate culprit around these if this is a bug. Debug your problem and let us know.
https://github.com/tmk/tmk_keyboard/blob/master/converter/ibm4704_usb/keymap_common.h#L136
https://github.com/tmk/tmk_keyboard/blob/master/converter/ibm4704_usb/matrix.c#L87-L109

NFiorentini, 2017-04-26:
> With default firmware the key spits out F16, couldn't you see F16 with tool like aquakey or switchhitter?

With hid_listen, that key spits out FB/7B, regardless of whether I use the [default](http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibm4704_usb_rev1_alps) or a [custom](https://goo.gl/59RJdS) firmware.

>It is likely to locate culprit around these if this is a bug. Debug your problem and let us know.

I don't know how to debug C.

We've got a discussion going on [here](https://deskthority.net/workshop-f7/converting-an-ibm-pingmaster-t16109-90.html).  There is at least one other person who is experiencing some or all of the previously-mentioned problems.  So far, the things in common between us seem to be using a Teensy2.0, the Teensy loader, and getting our .hex file from the TMK Keyboard Editor.   People who have compiled their firmware on their machine do not seem to be having this issue.

(I have been unable to compile firmware on my Windows machine and can't seem to get it to work.)

tmk, 2017-04-26:
hmm, binaries might be old and rotten somehow. Updated binaries, try them.

If you are using Keymap Editor you will have to wait for github to renew 
repo and reload page few times in your brower to download new binary.

NFiorentini, 2017-04-26:
Interesting.  Assigning something to the key _below_ the problem key re-assigns that problem key.  For example, using [this](https://goo.gl/DoXy2J) layout, that last Row 4 key is now Page Down.  And if I change that key below the problem key to "A," the problem key types "A." 

With [this](https://goo.gl/yhcYbT) layout, the problem key is Page Down and the one below it is Page Up.  Flipped.

Thank you for checking into this!

tmk, 2017-05-03:
`Keyapd Enter` and `Keypad =`(in default keymap) on right bottom were swaped in keymap editor. Fxied this at: https://github.com/tmk/tmk_keyboard/commit/de0aa0536ad691ff3f01747b4ee0e3607d0f2a63

I think this resolves your issue.

closed 2017-05-03 by tmk

NFiorentini, 2017-05-12:
Thank you!  That did resolve the flipped key issue.  And I haven't had any keys cease functioning since the binaries were updated.  Thank you!

However, I still have the issue of the board being non-responsive on the initial keypress if it has been idle for a moment or so.  I don't think this is an issue with my system because it only happens on this keyboard when using the TMK editor, though I could be (and am often) wrong.

Is there something in the TMK firmware that could cause this issue?  And do you have any tips for diagnosing this issue?

Thank you again for all you've done (and still do!) for the mech keyboard community!