tmk/tmk_keyboard #468: I wrote an improvement for anti-ghosting, works on all keyboards

barrar, 2017-05-13:
With anti ghost off, pressing K60, K61 and K50 works fine, but turned on it detects a ghost on K51. On my keymap there is no real K51 key connected to the matrix. This spot in the matrix is marked KC_NO.

I generate an array of real keys and empty spots in matrix:
static matrix_row_t matrix_ghost_check[MATRIX_ROWS] = {
0b001101111110111,
0b101111111101110,
0b010010111111111,
0b111111001111001,
0b111011101111111,
0b111111111111001,
0b110110110110111,
0b111101010011110
};

You can see final code here, sorry I can't just make a pull request because I changed many other files. It's just a small addition if you want to copy paste from this:

https://github.com/barrar/tmk_keyboard/commit/83ac2f3f18304e81c80f0e8a5fa9df120c4298bb

tmk, 2017-05-13:
not sure what your problem is. why don't you just keep the anti ghost off?
how did you turn the anti ghost  on/off? does your keyboard matrix has diodes?

barrar, 2017-05-13:
Keyboard does not have diodes. This is a finished feature addition. Please see link to code.

tmk, 2017-05-13:
ah, ok . I think I understand your idea. interesting.

closed 2017-05-13 by tmk

barrar, 2017-05-13:
I actually just improved it, here is the copy paste

```
#ifdef MATRIX_HAS_GHOST
extern const uint8_t keymaps[][MATRIX_ROWS][MATRIX_COLS];
// bit map of true keys and empty spots in matrix, each row is reversed
static uint16_t get_row_ghost_check(uint8_t row){
    uint16_t output = 0;
    for (int col = 0; col < MATRIX_COLS; col++) {
        if (keymaps[0][row][col] & 0xFF)
            output |= 1<<col;
        else
            output |= 0<<col;
    }
    return output;
}
static bool has_ghost_in_row(uint8_t row)
{
    matrix_row_t matrix_row = (matrix_get_row(row) & get_row_ghost_check(row));
    /* No ghost exists when less than 2 keys are down on the row.
    If there are "active" blanks in the matrix, the key can't be pressed by the user,
    there is no doubt as to which keys are really being pressed.
    The ghosts will be ignored, they are KC_NO.   */
    if (((matrix_row - 1) & matrix_row) == 0){
        return false;
    }
    // Ghost occurs when the row shares column line with other row, blanks in the matrix don't matter
    // If there are two or more real keys pressed and they match another row's real keys, the row will be ignored.
    for (uint8_t i=0; i < MATRIX_ROWS; i++) {
        if (i != row && __builtin_popcount((matrix_get_row(i) & get_row_ghost_check(i)) & matrix_row) > 1){
            return true;
        }
    }
    return false;
}
#endif
```