tmk/tmk_keyboard #473: Is possible make a 1x16 pad without a diode matrix?

YCF, 2017-05-28:
I'm making a 4x4 numpad with 32u4.As we know,32u4 have enough io to connect each key.
so,How to configure the matrix.c ?

Inspired by:
https://www.reddit.com/r/MechanicalKeyboards/comments/66588f/wireless_split_qmk_mitosis/dggoepg/

tmk, 2017-05-28:
Just check 16 pins and return their state.  You don't have to have row line.

YCF, 2017-05-28:
row to GND or VCC,which is better?

tmk, 2017-05-28:
GND.

ah, i checked  the link now, if low power consumption matters for this  I think you better have a pin for the row and make high when idle.

YCF, 2017-05-28:
4x4 matrix
vs 
16 io

how about the CPU utility？

tmk, 2017-05-28:
i guess 1x16 is lighter in cpu load.

YCF, 2017-05-28:
I don't care  about the power matter，I just  want to kill the diodes,so we don't have to solder it.
;p

closed 2017-05-28 by YCF