tmk/tmk_keyboard #483: Utilizing backlight function

undercookedrice, 2017-07-22:
Looking for some help in implementing the backlight code in gh60 to control multiple LEDs wired in parallel to toggle on/off, decrease brightness and increase brightness. I am not entirely sure how to set this up as there are no updated examples but I tried adding a backlight_set function to backlight.c in common, containing the pin that the LEDs are wired to (similar to the way led.c is set up). I also have the ACTION_BACKLIGHT_... functions mapped to different Fn keys in keymap_poker.c. Tried numerous things but kept getting errors like _undefined reference to eeconfig_. I noticed that some functions in the eeconfig files were not defined but did not know how to fix this.

kekstee, 2017-07-24:
You'd have to check whether the LED pins used on the GH60 can use a counter on chip. In that case you can look at my keyboard/nerd/backlight code for an example on how to set brightness (all at once, you need to modify tmk core for separate brightness values in eeprom)

Also my gh60 code does cover the easy cases of just setting the different leds.

You can check my changes to the gh60 and nerd keyboards here:
https://github.com/kekstee/tmk_keyboard

undercookedrice, 2017-07-24:
Thanks! I'll look into it :)

closed 2017-08-18 by undercookedrice