tmk/tmk_keyboard #484: Error while compiling One key teensy lc on a Teensy 3.2

4sStylZ, 2017-08-15:
Hi all,

I am new on Teensy and I have this error while compiling the teensy_lc_onekey keyboard : 

```
asstylz@elementare:~/Projects/tmk_keyboard/keyboard/teensy_lc_onekey$ make -f Makefile.3.2
Linking build/ch.elf
build/obj/eeconfig.o: In function `eeprom_initialize':
/home/asstylz/Projects/tmk_keyboard/keyboard/teensy_lc_onekey/../../tmk_core/common/chibios/eeconfig.c:330: undefined reference to `__eeprom_workarea_start__'
/home/asstylz/Projects/tmk_keyboard/keyboard/teensy_lc_onekey/../../tmk_core/common/chibios/eeconfig.c:330: undefined reference to `__eeprom_workarea_end__'
/home/asstylz/Projects/tmk_keyboard/keyboard/teensy_lc_onekey/../../tmk_core/common/chibios/eeconfig.c:330: undefined reference to `__eeprom_workarea_end__'
/home/asstylz/Projects/tmk_keyboard/keyboard/teensy_lc_onekey/../../tmk_core/common/chibios/eeconfig.c:330: undefined reference to `__eeprom_workarea_start__'
collect2: error: ld returned 1 exit status
../../tmk_core/tool/chibios/ChibiOS/os/common/startup/ARMCMx/compilers/GCC/rules.mk:258 : la recette pour la cible « build/ch.elf » a échouée
make: *** [build/ch.elf] Erreur 1
```

Sorry it’s a french Linux. Échoué means « Failed ».

I will try to provide you the most information than I can : 

* I run all of that on a ElementaryOS (based on Ubuntu).
* I have installed Arduino Ide, Teensyduino, Teensy Loader Application.
* I have moved the Teensy rules like described on the Teensyduino install process.
* I have cloned TMK.
* I have installed arm embedded GCC with this link : launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa In fact

This is the link provided for Ubuntu users and I have used thoses command : 

```
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
sudo apt-get install gcc-arm-embedded
```

The `arm-none-eabi-gcc -v` work well but I’l have some error code when installing « gcc-arm-embedded » so I am not sure that this is well installed.

I have cloned the two ChibiOS projects on the folder into this folder `/home/asstylz/Projects/tmk_keyboard/tmk_core/tool/chibios/`

tmk, 2017-08-17:
I'm not sure what is actual problem.
Gcc-arm I'm using on ubuntu is 4.9.3 and which can compile the sources.

4sStylZ, 2017-08-17:
Thank’s for the reply.

*PS : I am new at all the stack so sorry if my way of think is totally stupid.*

Hi have another installation of ElementaryOS (same version than my other computer who don’t compile). Hi have installed differently the gcc arm tooltchain simply via too standalone universe package : 

* `gcc-arm-none-eabi`
* `libnewlib-arm-none-eabi`

With this, the code compile. So I think that the different PPA provided in the instruction.md of Teensy_lc_one-key seems too cause my problem.

I will soon uninstall the gcc arm toolchain from my computer wich I have installed via this link https://launchpad.net/gcc-arm-embedded (This link is provided in the `installation.md` of the `teensy_lc_onekey`) and detailled here https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa if you follow the instruction for Ubuntu : 

> For Ubuntu 12.04/14.04/16.04/16.10 32/64-bit user, PPA is available at https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa.

The installation on Ubuntu with the instuctions on the ppa page seem’s not work (on Elementary and/or Ubuntu and/or for TMK or for another mystical reason)  : 

> Step1: Inside Ubuntu, open a terminal and input
>       "sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa"
> Step2: Continue to input
>       "sudo apt-get update"
> Step3: Continue to input to install toolchain
>       "sudo apt-get install gcc-arm-embedded"

After my fresh install I will update the ticket if I can help another person who have the issue.

Nosreme, 2017-11-01:
Hello,

I am running into the exact same issue on a mac.  I get the same issue on the gcc-arm 6-2017-q1-update and the 6-2017-q2-update.  This issue only occurs when I attempt to use the 3.2 or 3.0 makefile.  The lc makefile is fine.  

Any idea as to what could be causing this?

4sStylZ, 2017-11-23:
Hi. I haver never used a Mac in my life so I can’t help you.

If the package are specifics for Mac OSX I recommand you to create a dedicated issue with many information about your problem.

People generally don’t read the comments in a closed issue so speak here isn’t the best way for that.
Good luck.